rust-osdev/bootloader #5: Can't build the code

phodina, 2018-03-17:
Building the bootloader fails during the linking stage. It appears some funcs are not implemented in rlibc.
 
```
error: linking with `cc` failed: exit code: 1
  |
  = note: "cc" "-Wl,--as-needed" "-Wl,-z,noexecstack" "-m64" "-L" "/home/cylon2p0/.xargo/HOST/lib/rustlib/x86_64-unknown-linux-gnu/lib" "/home/cylon2p0/bootloader/target/debug/deps/bootloader-135935f7d9f3d13c.114xigf1yyw2gcw0.rcgu.o" "/home/cylon2p0/bootloader/target/debug/deps/bootloader-135935f7d9f3d13c.16u6js6g0l3k1ic6.rcgu.o" "/home/cylon2p0/bootloader/target/debug/deps/bootloader-135935f7d9f3d13c.170lmw1sqqfe80qo.rcgu.o" "/home/cylon2p0/bootloader/target/debug/deps/bootloader-135935f7d9f3d13c.1dk5kk1ooqmqppyo.rcgu.o" "/home/cylon2p0/bootloader/target/debug/deps/bootloader-135935f7d9f3d13c.1dqvxks6k2bzkxe.rcgu.o" "/home/cylon2p0/bootloader/target/debug/deps/bootloader-135935f7d9f3d13c.1dtnwzxvq606tnr.rcgu.o" "/home/cylon2p0/bootloader/target/debug/deps/bootloader-135935f7d9f3d13c.1gwav539dtqo5mpz.rcgu.o" "/home/cylon2p0/bootloader/target/debug/deps/bootloader-135935f7d9f3d13c.1jlhl2cwhfjqk73.rcgu.o" "/home/cylon2p0/bootloader/target/debug/deps/bootloader-135935f7d9f3d13c.1rumt9cqnarag56v.rcgu.o" "/home/cylon2p0/bootloader/target/debug/deps/bootloader-135935f7d9f3d13c.1y0f1d2ywo4h8fck.rcgu.o" "/home/cylon2p0/bootloader/target/debug/deps/bootloader-135935f7d9f3d13c.1y16o1qfye96o7m0.rcgu.o" "/home/cylon2p0/bootloader/target/debug/deps/bootloader-135935f7d9f3d13c.236iz806sghm3wx.rcgu.o" "/home/cylon2p0/bootloader/target/debug/deps/bootloader-135935f7d9f3d13c.27e9glo77wiztta4.rcgu.o" "/home/cylon2p0/bootloader/target/debug/deps/bootloader-135935f7d9f3d13c.2jqywn86b2gsqohu.rcgu.o" "/home/cylon2p0/bootloader/target/debug/deps/bootloader-135935f7d9f3d13c.2lyh15q6cjwzy18c.rcgu.o" "/home/cylon2p0/bootloader/target/debug/deps/bootloader-135935f7d9f3d13c.2qnu5lz8itx5t8lt.rcgu.o" "/home/cylon2p0/bootloader/target/debug/deps/bootloader-135935f7d9f3d13c.2uoyeb1ud6wb16nh.rcgu.o" "/home/cylon2p0/bootloader/target/debug/deps/bootloader-135935f7d9f3d13c.340hcms0rqmw1pjh.rcgu.o" "/home/cylon2p0/bootloader/target/debug/deps/bootloader-135935f7d9f3d13c.39fsqmmlmfbk11x5.rcgu.o" "/home/cylon2p0/bootloader/target/debug/deps/bootloader-135935f7d9f3d13c.3ab2w6r2t45b4bca.rcgu.o" "/home/cylon2p0/bootloader/target/debug/deps/bootloader-135935f7d9f3d13c.3ik0x0hz6l66cx38.rcgu.o" "/home/cylon2p0/bootloader/target/debug/deps/bootloader-135935f7d9f3d13c.3ldk0i2zxftngav8.rcgu.o" "/home/cylon2p0/bootloader/target/debug/deps/bootloader-135935f7d9f3d13c.3mf543h185mr1r9y.rcgu.o" "/home/cylon2p0/bootloader/target/debug/deps/bootloader-135935f7d9f3d13c.436dotimmrgzkwfa.rcgu.o" "/home/cylon2p0/bootloader/target/debug/deps/bootloader-135935f7d9f3d13c.43v6g0y2xsxoggnt.rcgu.o" "/home/cylon2p0/bootloader/target/debug/deps/bootloader-135935f7d9f3d13c.45pc7c65foh9i35f.rcgu.o" "/home/cylon2p0/bootloader/target/debug/deps/bootloader-135935f7d9f3d13c.48721dc4k5qxei0u.rcgu.o" "/home/cylon2p0/bootloader/target/debug/deps/bootloader-135935f7d9f3d13c.49a7n47po4ttqjl7.rcgu.o" "/home/cylon2p0/bootloader/target/debug/deps/bootloader-135935f7d9f3d13c.4h0kvux6xasoqj9f.rcgu.o" "/home/cylon2p0/bootloader/target/debug/deps/bootloader-135935f7d9f3d13c.4hqczukarqnsxim3.rcgu.o" "/home/cylon2p0/bootloader/target/debug/deps/bootloader-135935f7d9f3d13c.4re5h8xun1god8vx.rcgu.o" "/home/cylon2p0/bootloader/target/debug/deps/bootloader-135935f7d9f3d13c.4svjkysiwylw53vz.rcgu.o" "/home/cylon2p0/bootloader/target/debug/deps/bootloader-135935f7d9f3d13c.4tz198z0woys4rj8.rcgu.o" "/home/cylon2p0/bootloader/target/debug/deps/bootloader-135935f7d9f3d13c.4xq48u46a1pwiqn7.rcgu.o" "/home/cylon2p0/bootloader/target/debug/deps/bootloader-135935f7d9f3d13c.4yh8x2b62dcih00t.rcgu.o" "/home/cylon2p0/bootloader/target/debug/deps/bootloader-135935f7d9f3d13c.4ypvbwho0bu5tnww.rcgu.o" "/home/cylon2p0/bootloader/target/debug/deps/bootloader-135935f7d9f3d13c.51s1w397y42gpez1.rcgu.o" "/home/cylon2p0/bootloader/target/debug/deps/bootloader-135935f7d9f3d13c.56dhlykxk36fnj7k.rcgu.o" "/home/cylon2p0/bootloader/target/debug/deps/bootloader-135935f7d9f3d13c.57k06xfugllsc526.rcgu.o" "/home/cylon2p0/bootloader/target/debug/deps/bootloader-135935f7d9f3d13c.594xacf7fal17zd5.rcgu.o" "/home/cylon2p0/bootloader/target/debug/deps/bootloader-135935f7d9f3d13c.5csepo1d58sdgir4.rcgu.o" "/home/cylon2p0/bootloader/target/debug/deps/bootloader-135935f7d9f3d13c.7p53qlves2rwxx0.rcgu.o" "/home/cylon2p0/bootloader/target/debug/deps/bootloader-135935f7d9f3d13c.8xzrsc1ux72v29j.rcgu.o" "/home/cylon2p0/bootloader/target/debug/deps/bootloader-135935f7d9f3d13c.98g0d9x8aw3akpe.rcgu.o" "/home/cylon2p0/bootloader/target/debug/deps/bootloader-135935f7d9f3d13c.9elsx31vb4it187.rcgu.o" "/home/cylon2p0/bootloader/target/debug/deps/bootloader-135935f7d9f3d13c.ehly403ieh2api9.rcgu.o" "-o" "/home/cylon2p0/bootloader/target/debug/deps/bootloader-135935f7d9f3d13c" "-Wl,--gc-sections" "-pie" "-Wl,-z,relro,-z,now" "-nodefaultlibs" "-L" "/home/cylon2p0/bootloader/target/debug/deps" "-L" "/home/cylon2p0/.xargo/HOST/lib/rustlib/x86_64-unknown-linux-gnu/lib" "-Wl,-Bstatic" "/home/cylon2p0/bootloader/target/debug/deps/libspin-c51b2e86c2de1538.rlib" "/home/cylon2p0/bootloader/target/debug/deps/libos_bootinfo-dba7815e5fe96d50.rlib" "/home/cylon2p0/bootloader/target/debug/deps/libarrayvec-53c961d979185dfa.rlib" "/home/cylon2p0/bootloader/target/debug/deps/libnodrop-5ac98ef241ef76ef.rlib" "/home/cylon2p0/bootloader/target/debug/deps/libx86_64-8aa0ca550f385699.rlib" "/home/cylon2p0/bootloader/target/debug/deps/libux-1c933097b4b69b23.rlib" "/home/cylon2p0/bootloader/target/debug/deps/libusize_conversions-5b96477c65312c01.rlib" "/home/cylon2p0/bootloader/target/debug/deps/libbit_field-f6dd7c703522e559.rlib" "/home/cylon2p0/bootloader/target/debug/deps/libbitflags-a50202da937bfed0.rlib" "/home/cylon2p0/bootloader/target/debug/deps/libxmas_elf-28f6da88f7f9c0cf.rlib" "/home/cylon2p0/bootloader/target/debug/deps/libzero-359b6889bcf3f9b5.rlib" "/home/cylon2p0/bootloader/target/debug/deps/librlibc-a70e38fe72970e76.rlib" "/home/cylon2p0/.xargo/HOST/lib/rustlib/x86_64-unknown-linux-gnu/lib/libcore-6b8006ed673e894b.rlib" "-Tlinker.ld" "-Wl,-Bdynamic"
  = note: /home/cylon2p0/bootloader/target/debug/deps/bootloader-135935f7d9f3d13c.170lmw1sqqfe80qo.rcgu.o: In function `_start':
          170lmw1sqqfe80qo-d74921964f837aa8fb1308e78eea466c.rs:(.boot+0x0): multiple definition of `_start'
          /usr/lib/gcc/x86_64-pc-linux-gnu/7.3.0/../../../../lib/Scrt1.o:(.text+0x0): first defined here
          /usr/lib/gcc/x86_64-pc-linux-gnu/7.3.0/../../../../lib/Scrt1.o: In function `_start':
          (.text+0x12): undefined reference to `__libc_csu_fini'
          (.text+0x19): undefined reference to `__libc_csu_init'
          (.text+0x20): undefined reference to `main'
          (.text+0x26): undefined reference to `__libc_start_main'
          collect2: error: ld returned 1 exit status

```

phil-opp, 2018-03-17:
It works on my machine and [on travis](https://travis-ci.org/rust-osdev/bootloader/builds/354771461?utm_source=github_status&utm_medium=notification). Which linker are you using? Maybe we need to add `-nostartfiles` to the `pre-link-args` field in the target specification.

phodina, 2018-03-17:
Thanks @phil-opp . Now it compiles correctly.

phil-opp, 2018-03-17:
With `-nostartfiles`?

phil-opp, 2018-03-18:
Ah, I just saw your edit. Perfect!

I think we can close this then.

closed 2018-03-18 by phil-opp