rust-osdev/bootloader #6: Failed to load second stage of bootloader

phodina, 2018-03-17:
The booloader fails to load the second stage when run in qemu. In func load_second_stage_from_disk in file boot.s the code jumps to label second_stage_load_failed as the carry flag has been set.

phil-opp, 2018-03-17:
You have to compile in release mode. Not sure why yet, but I assume that it grows too big otherwise.

phodina, 2018-03-17:
I tried the release mode as you suggested but it also failed to load second stage.

Commands:
```
RUST_TARGET_PATH=`pwd` xargo run --target x86_64-bootloader --release
objcopy -O binary -S target/x86_64-bootloader/release/bootloader target/bootloader.bin
qemu-system-x86_64 -hda target/bootloader.bin -d int -s
```

rybot666, 2018-08-26:
Also seems solved, can we close this

phil-opp, 2018-08-26:
Agreed, but we definitely need to provide documentation for this.

closed 2018-08-26 by phil-opp