rust-osdev/bootloader #8: ld.bfd not found

weixiao-huang, 2018-04-01:
I try to build the code in MacOS 10.13.3 by using

```
$ RUST_TARGET_PATH=$(pwd) xargo build --target x86_64-bootloader --release
```

However, I got the error below:

[image]

phil-opp, 2018-04-02:
The problem is that the LLVM linker `lld`, which is shipped with Rust, fails to link the bootloader due to some unsupported 16-bit relocations. Thus, we have to use the GNU linker `ld.bfd`, which is only available on Linux by default. For building on mac OS, you have to [cross compile `binutils`](https://os.phil-opp.com/cross-compile-binutils/) (which contains the linker) and adjust the linker name in the `x86_64-bootloader.json` file. 

We try to find a solution to this, but I'm not sure whether we can fix this from our side or whether we need to wait for a new LLD version.

Nickforall, 2018-04-30:
This might be worth mentioning in the README under the `Build and Run` section?

Also, if you don't want your macbook to die from cross-compiling binutils, I recommend [this homebrew tap](https://github.com/SergioBenitez/homebrew-osxct).

phil-opp, 2018-05-01:
@Nickforall 

> Also, if you don't want your macbook to die from cross-compiling binutils, I recommend this homebrew tap.

I'm not a Mac user and don't know much about homebrew, but if this tap is really easier/better than building manually we could link it on https://os.phil-opp.com/cross-compile-binutils/.

Nickforall, 2018-05-01:
Well, compiling binutils is a long and cpu intensive process. Especially when you have a low-end dual core macbook. It took me a few hours with a relatively new model.

[Homebrew](https://brew.sh/) is a really popular package manager for OSX, and almost every developer using a mac has Homebrew installed.

By the way, if you need any help with testing on Mac, feel free to @ me in github issues or on gitter.

phil-opp, 2018-05-01:
Oh, that's a really long time. And the tap includes prebuilt versions?

> By the way, if you need any help with testing on Mac, feel free to @ me in github issues or on gitter.

Thanks a lot!

Nickforall, 2018-05-01:
Yep, it installs a prebuilt binary

xacrimon, 2018-05-01:
@weixiao-huang The instructions on how to solve the issue is in an open PR ready to be merged. I think we can close this now @phil-opp

phil-opp, 2018-05-01:
#12 was merged, so I'm going to close this. Thanks for reporting @

closed 2018-05-01 by phil-opp