rust-osdev/bootloader #13: How does one get the multiboot information address?

JacobSinclair, 2018-05-06:
From your first volume, you would do something like
```assembly
start:
    mov %rsp, stack_top
    mov %rdi, %rbx
```
and then add `multiboot_information_address: usize` to the "kmain" (or in the case of volume 2, _start).

However, when I add 
```rust
global_asm!(r#"
start:
    mov %rsp, stack_top
    mov %rdi, %rbx
"#);
```
to the top of my main.rs, and do
```rust
pub extern "C" fn _start(multiboot_information_address: usize) -> ! {
    let boot_info = unsafe{ multiboot2::load(multiboot_information_address)};
    let memory_map_tag = boot_info.memory_map_tag()
        .expect("Memory map tag required");

    println!("memory areas:");
    for area in memory_map_tag.memory_areas() {
        println!("    start: 0x{:x}, length: 0x{:x}",
            area.base_addr, area.length);
    }
    ...
```
The kernel panics with 
[image]
(I added a function to handle panics)

Thank you in advance

JacobSinclair, 2018-05-06:
I just asked on the OSdev discord and they say it needs to be passed from the bootloader, does the bootloader pass this information and, if so, how would I work with it?

xacrimon, 2018-05-06:
The bootloader passes the information. Right in the beginning of your asm boot code, add this
```
mov esp, stack_top
mov edi, ebx
```

Then make your main rust function accept a argument with the usize type. Thats the pointer

robert-w-gries, 2018-05-06:
Hi Jacob, I'm actually writing a blog post about how to migrate to the new bootloader with an existing Multiboot kernel!  Here's what I did to get my kernel working with the Rust bootloader (you can see all of my changes to my kernel [here](https://github.com/robert-w-gries/rxinu/pull/60)):

See the [end of the bootloader/main.rs#load_elf() function](https://github.com/rust-osdev/bootloader/blob/master/src/main.rs#L191-L204):

```rust
    // Construct boot info structure.
    let mut boot_info = BootInfo::new(recursive_page_table_addr.start_address().as_u64(), memory_map);
    boot_info.memory_map.sort();

    // Write boot info to boot info page.
    let boot_info_addr = boot_info_page.start_address();
    unsafe { boot_info_addr.as_mut_ptr::<BootInfo>().write(boot_info) };

    // Make sure that the kernel respects the write-protection bits, even when in ring 0.
    enable_write_protect_bit();

    let entry_point = VirtAddr::new(entry_point);

    unsafe { context_switch(boot_info_addr, entry_point, stack_end) };
```

The bootloader creates a [`BootInfo`](https://github.com/rust-osdev/os_bootinfo/blob/master/src/lib.rs#L12-L16) object that stores the address of the bootloader-mapped `PageTable` and a `MemoryMap` object.

The bootloader then calls a `context_switch()` function with the address of our `boot_info` along with another two variables `entry_point` and `stack_end`.

[`context_switch()`](https://github.com/rust-osdev/bootloader/blob/master/src/context_switch.s#L4-L16) is a simple assembly routines that loads our stack pointer and jumps to the kernel entry point.

Note the comments above the routine:

```asm
# Expected arguments:
# - boot info ptr in `rdi`
# - entry point address in `rsi`
# - stack pointer in `rdx`
context_switch:

    # load stack pointer
    mov rsp, rdx

    # jump to entry point
    jmp rsi
context_switch_spin:
    jmp context_switch_spin
```

The routine expects `boot_info_address` to be stored in the `rdi` register.  And this `boot_info_address` remains in register `rdi` when we call the kernel entry point, in our case `_start()`.  Per [System V AMD64 calling conventions](https://en.wikipedia.org/wiki/X86_calling_conventions#System_V_AMD64_ABI), `boot_info_address` will be passed forward as the first argument to the `_start()` function in our kernel.

So your `_start()` function was already getting the `boot_info_address` information! There isn't a need to add any assembly.

However, I would suggest changing the name of the parameter in your `_start()` function because we do not use Multiboot anymore with the Rust bootloader:

```diff
-pub extern "C" fn _start(multiboot_information_address: usize) -> ! {
+pub extern "C" fn _start(boot_info_address: usize) -> ! {
```

The only things left to do are 1) Load the `boot_info` object and 2) Check the version of `os_bootinfo` to ensure the bootloader version of `os_bootinfo` matches the version of `os_bootinfo` used in our kernel.

```rust
let boot_info: &BootInfo = unsafe { &*(boot_info_address as *mut BootInfo) };
if boot_info.check_version().is_err() {
    panic!("os_bootinfo version passed by bootloader does not match crate version!");
}
```

Now you should have access to the mapped `PageTable`s and the `MemoryMap` of the kernel, both provided by the bootloader!  The next step is to revamp the memory module to use this information, which I am in the process of writing.

--------

@phil-opp One question, is it still necessary to use `check_version()`?

closed 2018-05-06 by JacobSinclair