rust-osdev/bootloader #14: Don't mask IRQs

phil-opp, 2018-05-22:
Currently the bootloader masks all hardware interrupts: https://github.com/rust-osdev/bootloader/blob/bda13d807d0c7451b1ffe74c6c97f6369aef6ff9/src/second_stage.s#L68

They shouldn't be masked when control is passed to the kernel.

robert-w-gries, 2018-05-29:
Hey Phil, are you saying that interrupts should be enabled when the bootloader jumps to `_start()`?

phil-opp, 2018-05-29:
No, enabling them would not be a good idea since there isn't an IDT yet. I mean that they shouldn't be _masked_. An interrupt can be masked in the interrupt controller, so that the controller does not emit such an interrupt, even when interrupts are generally enabled.

rybot666, 2018-08-25:
Since this was patched, can we close the PR?

phil-opp, 2018-08-25:
@rybot666 Yes, thanks for the hint!

closed 2018-08-25 by phil-opp