rust-osdev/bootloader #16: Explanation / Tutorial

nebrelbug, 2018-06-07:
I love the idea of a pure Rust bootloader, but I'd love to know how it works, and perhaps write my own. Are there some good resources for this? I'd love to see a Phil-Opp tutorial about writing your own bootloader...

phil-opp, 2018-06-07:
I definitely plan to write such a tutorial, I'm just a bit short on time at the moment.

Also, a lot of the current implementation is still in (inline) assembly. I think it should be possible to convert the majority of that to Rust too, I just need to figure out how to compile for the 16-bit real mode.

nebrelbug, 2018-06-07:
@phil-opp Cool, thanks! Are there any good external resources?

phil-opp, 2018-06-09:
There are https://wiki.osdev.org/Bootloader and https://wiki.osdev.org/Rolling_Your_Own_Bootloader. Also you can look at existing code, for example the bootloader of the redox project.

There are also some tutorials for creating some minimal bootloader in assembly. I don't have any links, but you should be able to find them via google.

nebrelbug, 2018-06-11:
Sweet, thanks!

timClicks, 2019-02-16:
@nebrelbug don't know if this question is still on your mind, but the [first edition of the IntermezzOS book](https://intermezzos.github.io/book/first-edition/) walks readers through the process of writing a custom bootloader

nebrelbug, 2019-02-16:
@timClicks thanks! I'll check it out

phil-opp, 2019-02-16:
> the [first edition of the IntermezzOS book](https://intermezzos.github.io/book/first-edition/) walks readers through the process of writing a custom bootloader

Note that it builds upon the GRUB bootloader, so I wouldn't call it a bootloader. It covers the switch from protected to long mode and the page table setup, but it does not cover things like loading the kernel from disk, querying a memory map from the BIOS, or switching from real mode to protected mode.

Glab487894, 2024-01-07:
> Also, a lot of the current implementation is still in (inline) assembly. I think it should be possible to convert the majority of that to Rust too, I just need to figure out how to compile for the 16-bit real mode.

Are you going to leave api the same as it looks like now, or it is going to be changed after the convert assembly to Rust?

phil-opp, 2024-01-14:
Since v0.11, there's only [one small assembly file](https://github.com/rust-osdev/bootloader/blob/751585c6fa7ad825f5d21b38c09d7154f4b8acb6/bios/boot_sector/src/boot.s) left, the rest is in Rust already (even the 16-bit code!). So no further API changes are planned right now.

phil-opp, 2024-01-14:
As this thread is about a bootloader tutorial, let me link my in-progress guide on creating an bare-metal UEFI application, which can be the base for an UEFI bootloader: https://github.com/phil-opp/blog_os/blob/8a41fd65bf71f2c91d5c55f143a016ae27641e99/blog/content/edition-3/posts/02-booting/uefi/index.md

I have decided against writing more detailed bootloader tutorials for now, because things are very complex and somewhat hacky. Before even starting to write such a tutorial, I would have to create a less complex bootloader implementation that is easier to explain. This would be a lot of work, and I think it's better to focus my time elsewhere (the upcoming third edition of the blog, for example).

closed 2024-01-14 by phil-opp