rust-osdev/bootloader #26: Hangs before kernel entry

strake, 2018-09-28:
It seems to be a regression between 0.2.0 and 0.2.1: 0.2.0 works fine but 0.2.1 and 0.3.0 both hang.

Tested on qemu-3.0.0 on linux 4.14.18 x86_64

josephlr, 2018-10-04:
I was also encountering this issue when using the bootloader crate. Interestingly enough, I don't seem to run into this issue when using the precompiled version of the bootloader. I'll try to look into this further. It may be an issue with newer versions of rustc.

Tested on:
Linux 4.18.11
Qemu 3.0.0

josephlr, 2018-10-04:
So I did some more digging and was unable to figure out why the precompiled versions work, but the versions built from source do not. I had the whole setup working a month ago, so something has changed since then. It isn't:
- `bootimage` (tried with 0.4.4 and 0.5.6)
- `bootloader` (tried with 0.2.0 and 0.3.2)
- qemu (tried with 2.11.1 and 3.0.0)

So that leaves rustc as the reason things are breaking. There's most likely a bug in `bootloader` that's being triggered only on newer compilers. It could also be some other build tool (like `cargo-xbuild` or the linker).

josephlr, 2018-10-04:
So this issues seems to be resolved by building all dependencies with the latest nightly (`rustc 1.31.0-nightly (5597ee8a6 2018-10-03)`)

I wish I knew what the problem was

phil-opp, 2018-10-04:
@josephlr Thanks for investigating! I have no idea why this would happen. We should propably add more debug output to the bootloader, so that we see where it hangs if this happens again.