rust-osdev/bootloader #33: Rewrite build system?

phil-opp, 2018-11-24:
Currently the build process is a bit strange:

- First, we build the kernel
- Then we convert the ELF file to a binary
- Then we append a 512 byte sized structure that contains the kernel length
- Finally we append the kernel

We could change it in the following way:

- We could create a build script that reads an `BOOTLOADER_KERNEL_PATH` environment variable, converts it to a binary blob and then back to an object file (e.g. `cargo objcopy -- -I binary -O elf64-x86-64 target/x86_64-blog_os/debug/blog_os blog_os_bin.o  --binary-architecture=i386:x86-64`)
- Rename the default .data section to .kernel
- The build script then passes the object file to the linker
- This means that we can directly access the kernel blob in the linker script and directly access start/end/size of the kernel

The new build process would be:

- Build the kernel
- Build the bootloader (with an environment variable set to the kernel path)
- Convert the bootloader ELF file to a binary

The big disadvantage of this is that we need to rebuild the bootloader for each kernel change, so maybe it is better to keep the current build system.

longwusha, 2019-03-10:
I think rebuild the bootloader for each kernel change is not a good idea

phil-opp, 2019-03-10:
It wouldn't really recompile the bootloader, but it would rerun the build script and the linker. I think this shouldn't add much overhead, but I'm not 100% sure.

commit e1c63303 (2019-04-01): Merge #51

51: Rewrite build system r=phil-opp a=phil-opp

- [x] Blocked on https://github.com/rust-lang/rust/pull/59351
- [x] Blocked on https://github.com/rust-lang/cargo/pull/6778

Fixes #33 
Fixes #48
PR #51 (2019-04-01): Rewrite build system
- [x] Blocked on https://github.com/rust-lang/rust/pull/59351
- [x] Blocked on https://github.com/rust-lang/cargo/pull/6778

Fixes #33 
Fixes #48
closed 2019-04-01 by bors[bot]