rust-osdev/bootloader #44: Hangs before second stage on a real machine, works fine in QEMU

johanhenriksson, 2019-02-03:
Im following the OS dev tutorial series and decided to try to run the kernel on a real machine, but it hangs indefinitely before the second stage of the bootloader.

It prints ``Booting... (first stage)`` and then hangs. No other output unfortunately.

I've tried building it in release mode (as suggested in a similar issue) but it doesn't make a difference. Everything works fine when I run it in QEMU, release or debug.

Any ideas? Would be cool to see it running on real hardware :)

phil-opp, 2019-02-03:
I cannot reproduce this unfortunately. For me it runs fine on real hardware.

It seems like the bootloader does something that does not work on your hardware, which is difficult to debug from my end. Could you try to do some debugging to find out on which instruction it hangs? The easiest way to find this out is:

- Clone this bootloader project.
- Replace the `bootloader = 0.3.0` dependency in your kernel with a path dependency: `bootloader = { path = "../bootloader" }` (use the path to directory where you cloned the bootloader). Now the kernel should use your local bootloader for building.
- Edit the `src/stage_1.s` assembly file to find out where the error occurs
  - The `Booting... (first stage)` message is printed in [line 26](https://github.com/rust-osdev/bootloader/blob/master/src/stage_1.s#L25-L26) so we know that the problem is on a later line.
  - Try to insert the following lines at various places in the file to print the `Booting... (first stage)` message a second time:
  
  ```
    push esi    # backup `esi` because we overwrite it
    push eax    # backup `eax` `real_mode_println` overwrites it
    lea si, boot_start_str    # load the address of the message into `si`
    call real_mode_println    # print the message a second time
    pop eax    # restore `eax`
    pop esi    # restore `esi`
  ```

- Move these lines around and try to isolate the problematic instruction this way. If you see the message twice, all preceding instructions worked. If you see the message only once, the problem is in a preceding instruction.

johanhenriksson, 2019-02-05:
Sure, I will give it a shot tonight

johanhenriksson, 2019-02-06:
Okay, so this is pretty confusing 😄  I tried to add a print after the ``enable_a20`` instructions on line 33. It initially caused a reboot loop, but I recalled I shouldn't use 32 bit registers immediately after boot (right?) so I changed the print code to:

```
push si    # changed register name
push ax    # changed register name
lea si, boot_start_str    # load the address of the message into `si`
call real_mode_println    # print the message a second time
pop ax    # changed register name
pop si    # changed register name
```

And surprisingly, it booted into the kernel! Quickly enough that I can't see any bootloader text printed. The funny thing is, if I remove this print call again, it hangs sometime after the first print like before.

I'd love to understand why this is happening but I'm at a loss. Could it be that these few extra instructions fixes some alignment problem?

phil-opp, 2019-02-07:
@johanhenriksson That's really strange. I updated the A20 check in https://github.com/rust-osdev/bootloader/commit/81c9761926be6cfa2c8843f531db59e754af605c (in a separate `boot-fix` branch). Could you try whether this works for you?

To use that branch, change your bootloader dependency to the following:

```toml
[dependencies]
bootloader = { git = "https://github.com/rust-osdev/bootloader.git", branch = "boot-fix" }
```

johnwstanford, 2019-02-08:
I cloned bootloader, then changed "Booting (first stage)" to "Booting (1st stage)" just to make sure I was using my local version.  That worked as expected.

I added another string constant:

```
boot_start_str: .asciz "Booting (1st stage)..."
debug_str: .asciz "Debug line"
error_str: .asciz "Error: "
```

Then tried to print it right after the first line, with the idea that I'd move "Debug line" forward until I saw one but not the other:

```
    # initialize stack
    mov sp, 0x7c00

    lea si, boot_start_str
    call real_mode_println

    lea si, debug_str
    call real_mode_println
```

However when I do this, I only see the first line.  I don't see the "Debug line" I added.  Does this mean the call to real_mode_println itself is causing the freeze or am I calling it wrong the second time?  Thanks!

phil-opp, 2019-02-08:
@johnwstanford Thanks! I looked over the `real_mode_println` function and saw that it tries to do addressing with a 32-bit register [here](https://github.com/rust-osdev/bootloader/blob/b1d741f74cbbae2f7286ac1eb9f43e68c3b07c73/src/stage_1.s#L128), which should not work. I think this is a copy&paste error. Could you change the `[esi]` to `[si]` and try again?

johanhenriksson, 2019-02-09:
The ``boot-fix`` branch seems to work for me 👍

phil-opp, 2019-02-09:
@johanhenriksson Great, thanks for testing! I will merge the branch and pusblish a new version.

johnwstanford, 2019-02-09:
The change you told me to try didn't work, but I pulled down merge #46 and that one works for me :) Thanks!

phil-opp, 2019-02-11:
@johnwstanford Awesome! It should be published on crates.io as version 0.3.13.

I think we can close this issue then.

phil-opp, 2019-02-11:
@johnwstanford 

> The change you told me to try didn't work

Just to clarify: It didn't fix the issue of booting on real hardware, or it didn't work at all? Because it seems like a bug to me and I would like to fix it even if it wasn't the cause of this issue.

closed 2019-02-11 by phil-opp

johnwstanford, 2019-02-12:
@phil-opp Yes, changing [esi] to [si] at line 128 (now line 133) didn't fix the problem for me.  I got the same behavior where it compiled and started booting, but got hung at "Booting (first stage)..."  However, I pulled merge #46 down, compiled it, and now it works on real hardware like it's supposed to, but it still has [esi] instead of [si] at line 133 of stage_1.s.  I tried two different types of hardware and it works on both.  If it's helpful, I could change [esi] to [si] and make sure it still works.  Thanks!

phil-opp, 2019-02-12:
@johnwstanford Thanks for testing!

> If it's helpful, I could change [esi] to [si] and make sure it still works.

That would be great! I'm pretty sure that it's a bug, but I don't want to break anything. Could you try depending on #47:

```toml
[dependencies]
bootloader = { git = "https://github.com/rust-osdev/bootloader.git", branch = "fix" }
```

johnwstanford, 2019-02-13:
No problem.  That branch works on the real hardware too.

phil-opp, 2019-02-13:
Perfect, thanks!