rust-osdev/bootloader #56: Build error: unexpected llvm-size line output

phil-opp, 2019-06-16:
Reported in https://github.com/phil-opp/blog_os/issues/403#issuecomment-502367797.

phil-opp, 2019-06-16:
@LiamTheProgrammer I pushed a debug version of the bootloader that prints additional information when this error occurs. Could you replace your bootloader dependency by:

```toml
bootloader = {git = "https://github.com/rust-osdev/bootloader.git", branch = "debug"}
```

And then try to run your project again?

The error should still occur, but this time it should show the full `llvm-size` output as part of the panic message. This should help us debug this issue.

LiamTheProgrammer, 2019-06-17:
@phil-opp I managed to fix my issue. I forgot how, but here are the commands that I used:
```
C:\Users\Administrator\LunarOS>cargo xbuild --release --target config.json
   Compiling LunarOS v0.1.0 (C:\Users\Administrator\LunarOS)
warning: crate `LunarOS` should have a snake case name
  |
  = note: #[warn(non_snake_case)] on by default
  = help: convert the identifier to snake case: `lunar_os`

    Finished release [optimized] target(s) in 1.01s

C:\Users\Administrator\LunarOS>cargo bootimage --target config.json
Building kernel
   Compiling LunarOS v0.1.0 (C:\Users\Administrator\LunarOS)
warning: crate `LunarOS` should have a snake case name
  |
  = note: #[warn(non_snake_case)] on by default
  = help: convert the identifier to snake case: `lunar_os`

    Finished dev [unoptimized + debuginfo] target(s) in 0.50s
Building bootloader
   Compiling bootloader v0.6.0 (C:\Users\Administrator\.cargo\registry\src\github.com-1ecc6299db9ec823\bootloader-0.6.0)
    Finished release [optimized + debuginfo] target(s) in 9.96s
```

phil-opp, 2019-06-18:
@LiamTheProgrammer Thanks for the update! Since we both can no longer reproduce this issue, I'm going to close it. Let me know if it occurs again, maybe we can debug it then.

closed 2019-06-18 by phil-opp

monarrk, 2020-11-12:
Hey, I'm now experiencing this bug. I'm on NetBSD 9.0 with cargo version `cargo 1.49.0-nightly (d5556aeb8 2020-11-04)`. When I try to build the bootloader, I get the error ` thread 'main' panicked at 'unexpected llvm-size line output', build.rs:155:39`. Do you have any ideas on what this could mean?

phil-opp, 2021-01-03:
@monarrk Sorry for the late reply! If you still have this issue, try the debug instructions I mentioned above: https://github.com/rust-osdev/bootloader/issues/56#issuecomment-502457219