rust-osdev/bootloader #70: 32bit support

bjorn3, 2019-07-31:
cc https://github.com/phil-opp/blog_os/issues/403#issuecomment-516832927.

@Mendess2526 let's continue the discussion here.

This will need src/stage_3.s to be adapted to stay in protected mode and to setup non-PAE page tables. It will then need at least changes to the rust part to handle non-PAE page tables.

ethindp, 2019-07-31:
I hope this is togglable with a feature flag. Just saying, I'm
personally not in favor of this idea.

64, 2019-07-31:
Yeah it would need to be a toggleable feature. We should support the multiboot2 protocol for this

bjorn3, 2019-07-31:
> I hope this is togglable with a feature flag.

Shouldnt it just work when it is compiled for `i686` instead of `x86_64`?

64, 2019-07-31:
Yeah that’s a good point. I think in the mean time it might be good to add some check in the build script that it’s being built for 64-bit mode, otherwise error out and point to this issue

phil-opp, 2019-08-01:
>  I think in the mean time it might be good to add some check in the build script that it’s being built for x86_64 (if that is even possible), otherwise error out and point to this issue

This should be possible through the `CARGO_CFG_TARGET_ARCH` [environment variable](https://doc.rust-lang.org/cargo/reference/environment-variables.html#environment-variables-cargo-sets-for-build-scripts), but only [when the crate is used as a library](https://github.com/rust-osdev/bootloader/blob/0ffbebd6cce97c7616f7b56a5fa01410fe28739e/build.rs#L1-L2). The binary compilation currently occurs for a fixed target specified [in the `Cargo.toml`](https://github.com/rust-osdev/bootloader/blob/0ffbebd6cce97c7616f7b56a5fa01410fe28739e/Cargo.toml#L46), but this could be easily changed by updating bootimage (e.g. by making the `package.metadata.bootloader.target` a (target_arch, target file name) map).

ethindp, 2019-08-23:
Personally I don't think we should support 32-bit at all. The only reason I can see this being required is running this on old hardware, and you have VMs for that. Now that you can chainload the bootloader (via grub, for example), you can (technically) run the system on "real" (virtual, but technically real) hardware. Its not physically real, but virtualized hardware, with a few changes here and there, is based off of at least a model of old hardware (i.e. the VMware BIOS implementation uses Phoenix BIOS, I think). 32-bit support overcomplicates a lot of things and would just bloat the code to unreasonable proportions.