rust-osdev/bootloader #74: Enable support for SMP Trampoline

vinaychandra, 2019-08-14:
The current entry point is hardcoded to `_start` (or the `entry_point!` macro). This provides a single entry point to the code.

When bootstrapping SMP, the new processor starts up at a given physical address in the IPI (Inter-Processor Interrupt) in 16 bit mode. This requires us to start at 16 bit code and enable 64 bit mode again. Please provide a way to define multiple "entry points" which defaults to the current `_start` but can also be used to bootstrap SMP.

64, 2019-08-14:
I’m getting near SMP in my own kernel, so I’ll probably investigate this a bit. It might be difficult to do in a platform agnostic way. For example, how should the page tables be set up?

ethindp, 2019-08-14:
I've got PCI going and am just confused on how I should manage some
kind of driver registration framework to pass the PCI devices off to.
Heh, I don't even have networking, but SMP and APIC would be nice.

vinaychandra, 2019-08-14:
@64 the way I understand it, we don't need to setup new page tables, etc. We can leave other processors potentially in the same place as the bootstrap processor. The pagetables are all setup for it anyway. So, we assume the same and start up the new processor into the function provided by the user. Any other changes will have to be done by the developer.

64, 2019-08-14:
You don’t need to set up page tables, but you need to load *something* into cr3 when switching to long mode.

vinaychandra, 2019-08-14:
Agreed, that's where i said we can use the same assumptions as the bootstrap processor. We can load with the same CR3 but point EIP to the new entry point rather than the old one.

64, 2019-08-14:
Booting up the APs is usually done fairly late in the boot process. Some people may have switched to a different p4 table in this time, so I think it would be better to have some way of setting the value at runtime. But that shouldn’t be too difficult.

ethindp, 2019-08-14:
Is there a way we can make a separate rust crate for SMP and allow the
kernel to do all the SMP handling, or is that only doable by the
bootloader/BIOS/UEFI? I don't fully understand SMP myself, but having
that kind of control seems apt for a kernel.

64, 2019-08-14:
The bootloader doesn’t need to be involved with SMP at all. What *could 
* be useful is a crate which can be used for interacting with the LAPIC / I/O APICs, since that’s mostly what SMP is about.

vinaychandra, 2019-08-14:
I agree. The crate provides entry point for 16 bit bootstrapping, not just boot loader logic which i wanted to reuse for SMP rather than creating another setup which does the exact same thing. That was my request as well. Just provide a way to start in 16 bit code and end in 64 bit rust which can potentially be used for SMP

vinaychandra, 2020-01-22:
@phil-opp , i have been trying to play with this but hit a roadblock. SMP's processors start in 16 bit mode at a page aligned address. I couldn't find a way to change the current bootloader code so that i can add an extra function that will be placed at a page aligned address on startup. I didn't want to go through copying of pages in kernel. It would be nice if the low level kernel loader supports that directly.

Can you make sure such a thing would be easier in your rewrite?

bjorn3, 2020-01-22:
Would using `global_asm!` combined with `.align 512` before the function asm work?

vinaychandra, 2020-01-22:
`align 4096` should work (we need page aligned) but we should also be guaranteed that the function will be in the first 1MB of ram because that is the limit of real mode. Does bootloader guarantee that?

bjorn3, 2020-01-22:
> align 4096 should work (we need page aligned)

oops, got confused with the old disk sector size :)

phil-opp, 2020-01-22:
@vinaychandra 

I think the best way to do implement something like this is to add a `smp_trampoline` field to the boot information that points to the physical address of the trampoline function. The function itself could be created in assembly in the real mode part of the bootloader along with the normal initialization code. I think this should be relatively easy to add so that we won't have to wait for the rewrite of the lower stages. Would this approach work for you?

vinaychandra, 2020-01-22:
@phil-opp it should, as long as the final address of the function is page aligned, in first mb and we know the value in kernel.

phil-opp, 2020-01-23:
@vinaychandra Ok, sounds good. I don't have time to provide the full implementation right now unfortunately. How about I start with the boilerplate (adding a page aligned function in the first mb and exporting that address in the boot information) and let you (or others) create the real trampoline function?

vinaychandra, 2020-01-23:
Yes, that would be perfect. I can help out in creating the real trampoline (my first time but still interested)

phil-opp, 2020-02-19:
I created https://github.com/rust-osdev/bootloader/pull/97 with the required boilerplate. I hope this helps!

On a side note: Why the page alignment is required? I tried to google it, but didn't find anything…

vinaychandra, 2020-02-19:
Example from [this](https://stackoverflow.com/questions/48763883/apic-multi-core-startup-protocol-and-icr-starting-up-address) answer, 

```
A SIPI contains a vector, this is similar in meaning, but absolutely different in practice, to an interrupt vector (a.k.a. interrupt number).
The vector is an 8 bit number, of value V (represented as vv in base 16), that makes the CPU starts executing instructions at the physical address 0vv000h.
We will call 0vv000h the Wake-up address (WA).
The WA is forced at a 4KiB (or page) boundary.
```

When a processor starts up, you can send it a "startup" address which it will offset with 000h which is page aligned.

Andy-Python-Programmer, 2021-04-13:
Any updates on this?

Lucky4Luuk, 2023-03-01:
Any updates on this? This would be a really nice addition for this crate.

jasoncouture, 2023-03-05:
Copy/Paste from: https://github.com/rust-osdev/bootloader/issues/343#issuecomment-1454941913

> I, personally, don't think this is a good idea.
> 
> Here's why:
> 
> The bootloader already has a lot to do, with limited support and system resources (Especially in the case of bios). There are also different gotchas with various hardware iterations stretching back to the initial MP spec by intel. (IE: It's hard to get right across a variety of hardware). Beyond that, it requires reading and parsing the ACPI tables, and setting up the APIC to do IPI. Could it be done? Yes. But it would be less of a bootloader, and more of a kernel. It's also more information the kernel needs to take into account, how IPI was setup as an example.
> 
> Perhaps a better way to go about this would be to provide a no_std library that made it easy to start CPUs instead, and people could drop that into their kernel. Similar things exist for ACPI, Serial, and a few more things.

jzbor, 2023-03-05:
I think it would be nice to provide a method to _place_ the trampoline code through the bootloader, while leaving the whole IPI shenanigans to the actual OS. This would keep the implementation for the bootloader relatively simple while allowing the OS to rely on the memory management provided by this crate.

I tried (re-)implementing the SMP-bootcode from another project with a blogOS-like OS and the bootloader crate and could not get around the issue on where to put the bootcode without interfering with the bootloader/OS and without resorting to external tools for linking. I am sure that was a skill issue on my side, but I think it would help greatly if the bootloader would provide some space to put the trampoline code...

jzbor, 2024-02-23:
Has anyone succeeded in bringing up SMP with Cargo + bootloader crate?

jasoncouture, 2024-02-23:
SMP is outside of the scope of a bootloader. That said, the bootloader currently chews up the lower 1MB, which is the only place the AP trampoline can exist.

I made some modifications a while ago before getting sick that prevented the bootloader from using the lower 1mb of ram, and I was able to pull up the APs after I did that.

You can also use a linker script to place the trampoline, but due to system variations, it's best to do this at runtime.

jasoncouture, 2024-02-23:
> Has anyone succeeded in bringing up SMP with Cargo + bootloader crate?

Yes, see my previous comment.