rust-osdev/bootloader #75: Configuration does not work

vinaychandra, 2019-08-15:
I am slightly new to rust. Apologize for my mistake.
I am trying to add `[package.metadata.bootloader]` but it does not take effect.

Steps for repro:
1. Checkout `post-10` from blog_os
1. Add `println!("{:X}", boot_info.physical_memory_offset);` to `kernel_main`
1. Add the following snippet to cargo.toml at the end
```
[package.metadata.bootloader]
kernel-stack-address = "0xFFFFFF8000000000"
kernel-stack-size = 128
physical-memory-offset = "0xFFFF800000000000"
```

Expected output: `0xFFFF800000000000`
Current output: not `0xFFFF800000000000`

Am I missing something? Should I have to build bootloader myself to use the configuration?

64, 2019-08-15:
Are you using the latest version of the bootloader crate?

vinaychandra, 2019-08-15:
Had to update to "0.8" of bootloader.

closed 2019-08-15 by vinaychandra