rust-osdev/bootloader #76: Compiling with "binary" feature with example kernel fails

ethindp, 2019-08-23:
When building the bootloader as a dependency (i.e. to enjoy the extra features) the build fails:
```
thread 'main' panicked at 'The bootloader must be compiled for the `x86_64-bootloader.json` target.', C:\Users\ethin\.cargo\registry\src\github.com-1ecc6299db9ec823\bootloader-0.8.0\build.rs:108:9
```
I am attempting to take advantage of the dynamic stack selection, dynamic kernel placement, etc., that I would not otherwise enjoy had I not specified the "binary" feature. Is this doable?

phil-opp, 2019-08-23:
The `binary` feature should only be specified if you compile the bootloader as a binary. if it is a dependency in your Cargo.toml, you compile it as a library and should not add the feature.

The `bootimage` tool automatically sets the `binary` feature when it compiles the bootloader binary. Just use the latest bootloader version and you get all the benefits you mentioned.

phil-opp, 2019-09-18:
Seems to be answered. If this is still a problem for you, let us know!

closed 2019-09-18 by phil-opp