rust-osdev/bootloader #78: Build successfully with benign errors in WSL

Barronli, 2019-08-28:
When I build the bootloader under WSL (Windows subsystem Linux), it finishes while reporting errors:

>    Compiling bootloader v0.8.0 (/mnt/d/devs/Rust/bootloader)
> error: failed to remove /mnt/d/devs/Rust/bootloader/target/x86_64-bootloader/release/deps/bootloader-589b595378e59014.bootloader.3pdwijpa-cgu.0.rcgu.o: Input/output error (os error 5)
> 
> error: failed to remove /mnt/d/devs/Rust/bootloader/target/x86_64-bootloader/release/deps/bootloader-589b595378e59014.bootloader.3pdwijpa-cgu.1.rcgu.o: Input/output error (os error 5)
> 
> error: failed to remove /mnt/d/devs/Rust/bootloader/target/x86_64-bootloader/release/deps/bootloader-589b595378e59014.bootloader.3pdwijpa-cgu.12.rcgu.o: Input/output error (os error 5)
> 
> error: aborting due to 3 previous errors
> 
> Finished release [optimized + debuginfo] target(s) in 3.83s
> 

1. It does not report the errors if I enter the build command again. 
2. The mentioned files can be removed manually since they have all the r/w file permissions set. 
3. The same errors reported when building blog_os as well, and does not block anything. I.e., blog_os can run/test in WSL as expected.

phil-opp, 2019-08-29:
I think os error 5 is an "access denied" error. I'm not sure what exactly causes this, but it looks like rustc itself causes this issue when trying to remove old compilation artifacts.

Maybe it's related to https://github.com/rust-lang/rust/issues/48700?

Barronli, 2019-08-29:
@phil-opp  You are right, and it is a rustc related issue. I saw the evidence in the stack trace of codegen when it tries to remove the intermediate objects after linking. But this issue is different from the one you mentioned in rust-lang/rust#48700, since I am not using VScode. I suspected it was due to my antivirus software, but the problem persists after I disabled the file scanning. Anyway, I am closing this issue, and will open one in rust-lang. 
Btw, the debug build (without passing "--release") to cargo does not have the issue.

closed 2019-08-29 by Barronli

phil-opp, 2019-08-29:
Thanks for the update! Let me know if there is anything that I can do in this project to prevent or work around this issue.

Barronli, 2019-08-31:
It looks like if I build the project without the "--release" option, it does not report error, but I don't know how to force rustc (or cargo) to work the same way when with the "--release" option. I tried adding the following two lines (both and either) in Cargo.toml but got same error. I guess there are more differences between the release and dev profiles that I failed to find out by googling.
````
 [profile.release] 
  panic = "abort"
  lto = false
+ debug = true
+ opt-level = 0
````

For blog_os purpose, don't know if it is possible to set the dependency on bootloader's debug build. I tried to use the following cargo feature in blog_os Cargo.toml but failed too.

````
cargo-features = ["profile-overrides"]

[profile.release.overrides.bootloader]
debug = true
opt-level = 0
````
Then I realized that bootimage itself actually builds the kernel and bootloader too, and always builds the bootloader in release mode. I commented the line of ["--release"](https://github.com/rust-osdev/bootimage/blob/736ca1bf571f4745e49606017d6cf4e4216605d7/src/builder.rs#L233) in bootimage builder.rs, then I can build blog_os without the said errors, but I cannot run the OS correctly. I am afraid the --release build of bootloader is necessary?

phil-opp, 2019-08-31:
Yeah, it seems like `--release` is currently necessary for some reason, but I'm not sure why. I suspect that is has something to do with the assembly code that loads the rest of the bootloader. Either way, this issue doesn't sound like it would be really fixed with a debug build. Even if it doesn't happen for this crate and compiler version, it might happen again with different versions.