rust-osdev/bootloader #87: PXE support (with iPXE)

GuillaumeDIDIER, 2019-12-17:
Would it be possible to have support to boot using PXE or iPXE (with iPXE downloading the bootimage.bin using TFTP for instance).

This would require changing the logic to load the files in memory (which iPXE probably takes care of).

This could be poretty helpful for development on real hardware, as it wuld allow to network boot instead of going through the write to USB manipulations.

GuillaumeDIDIER, 2019-12-17:
gamozolabs/orange_slice may be an intersting ressource given that it has a PXE bootloader.

phil-opp, 2019-12-18:
This would be definitely useful. However, I don't have the time to look into this right now, as there are other roadmap items with a higher priority (support for multiboot and UEFI).

If someone wants to work on this, I would be happy to provide mentoring for the bootloader integration.

ethindp, 2021-07-11:
Shouldn't this just work out of the box? Considering that the bootloader is linked with the kernel, no intermediate steps should be necessary other than flashing the iPXE ROM and setting up an iPXE boot script to download the binary.

Freax13, 2022-05-10:
I've opened a pr that implements PXE support for the UEFI bootloader.

Is there any demand for implementing Legacy PXE support? I have been working on a simple implementation for a Legacy PXE  (see https://github.com/Freax13/bootloader/tree/legacy-pxe). If there's not much demand for it I'm not sure if it's worth integrating: Most newer devices don't implement Legacy PXE anymore, the code's not that pretty (mostly copied from the BIOS bootloader, but different in a few places that make it hard to split out the common parts) and the only platform I have to test behaves wildly different than QEMU.

ChocolateLoverRaj, 2025-03-09:
Btw if you don't want to deal with TFTP (like me) you can use iPXE with http to load this bootloader. With something like this:
```
#!ipxe
module http://gaming-computer:64017/kernel-x86_64
module http://gaming-computer:64017/ramdisk
chain http://gaming-computer:64017/efi/boot/bootx64.efi
```
but I will probably just use http since iPXE doesn't support modern certificate algorithms.