rust-osdev/bootloader #88: Recursive Index in page tables?

mark-i-m, 2019-12-31:
Is it always index `0`? Should this be passed in the `BootInfo` struct?

phil-opp, 2020-01-02:
We used to hardcode the index to `511`, but this changed with #71. Now we dynamically look for a free level 4 entry. The chosen index is indirectly reported through the `recursive_page_table_addr` field in the `BootInfo`:

https://github.com/rust-osdev/bootloader/blob/90f5b8910d146d6d489b70a6341d778253663cfa/src/bootinfo/mod.rs#L32-L34

It shouldn't be `0`, however, because the bootloader maps itself to that slot…

mark-i-m, 2020-01-02:
Thanks! I eventually figured that out. Perhaps a "get recursive index" method could be added that just masks out bits 21:12 and returns a PageTableIndex?

phil-opp, 2020-01-27:
Sure, I'd be happy to merge a pull request for this.

PR #116 (2020-05-22): Add recursive_idx for boot info
Closes #88 

Sorry, it's been a while!
closed 2020-05-22 by phil-opp