rust-osdev/bootloader #89: Documentation on initial state

mark-i-m, 2019-12-31:
Is there any documentation about the state of the system when `_start` function is invoked by the bootloader? For example, what features are enabled, what is the layout of memory, what state are the processors in?

phil-opp, 2020-01-02:
Not yet. I'm currently working on rewriting the first stages of the bootloader in Rust and plan to add more documentation in the process.

Do you have questions about a specific feature/processor register? Maybe I can answer them here.

mark-i-m, 2020-01-02:
Thanks for the response. I was able to figure out the following for the recursive mapping mode, but perhaps it would be good to document:

- (I believe) only the main processor is initialized. If you want to turn on SMP, you need to do it in your kernel.
- The bootloader loads itself at the beginning of virtual and physical memory (the first 16kb iirc)
- The vga buffer is direct mapped 0xb8000
- The kernel is mapped at Virtual address 0x200000 but it's backed by dynamically choose physical memory.
- Page tables are recursively mapped at a dynamic entry, and the recursive index can be extracted from the address in BootInfo.
- The BootInfo memory map gives all the physical regions used, including ones allocated by the bootloader.

phil-opp, 2020-01-27:
Seems correct to me, except for:

> The kernel is mapped at Virtual address 0x200000 but it's backed by dynamically choose physical memory.

The kernel is mapped by the virtual address specified by the ELF file. By default, the linker uses address `0x200000` for executables, but you can change that address using linker flags or a linker script.

> perhaps it would be good to document

Absolutely! A good start would be to create an `doc/environment.md` file with your bullet points.

vinaychandra, 2020-02-09:
@phil-opp , couple of questions
- How much of the actual physical memory is mapped when using offset mode? It is exactly 100% or more? (For example, what happens if i access `offset + 100GB` when I don't have 100GB of memory?
- By default, some areas of the memory are dedicated to memory mapped registers. (Higher regions of memory) How do you want to represent that in the memory map provided?

phil-opp, 2020-02-10:
The memory map is a [E820 memory map](https://wiki.osdev.org/Detecting_Memory_(x86)#BIOS_Function:_INT_0x15.2C_EAX_.3D_0xE820) created by the BIOS. The bootloader crate just removes any used memory from it, otherwise it is passed to the kernel with no modifications. From the top of my head, I'm not sure whether memory mapped registers are listed in it, but I don't think so.

The [code for mapping physical memory](https://github.com/rust-osdev/bootloader/blob/cc206c3b5cf6282d445e2699520482a984702b11/src/main.rs#L284-L322) is quite simple: It linearly maps all physical pages starting from 0 up to the largest physical address [reported in the memory map](https://github.com/rust-osdev/bootloader/blob/cc206c3b5cf6282d445e2699520482a984702b11/src/main.rs#L137-L141).

I hope this answers your questions!