rust-osdev/bootloader #90: `error: This crate currently does not support 32-bit protected mode` when compiling kernel with x86_64-unknown-none target

0x00AE6150, 2020-01-06:
After updating `bootloader`  to 0.8.3 I started getting `error: This crate currently does not support 32-bit protected mode` error but I am pretty sure that I cross-compile x86_64 arch binary.
This issue is not reproducible with bootloader version 0.7.1 or lower. Here is the content of my target x86_64-unknown-none.json:
{
  "llvm-target": "x86_64-unknown-none",
  "data-layout": "e-m:e-i64:64-f80:128-n8:16:32:64-S128",
  "arch": "x86_64",
  "target-endian": "little",
  "target-pointer-width": "64",
  "target-c-int-width": "32",
  "os": "none",
  "executables": true,
  "linker-flavor": "ld.lld",
  "linker": "rust-lld",
  "panic-strategy": "abort",
  "disable-redzone": true,
  "features": "-mmx,-sse,+soft-float"
}

I am compiling with this command:
cargo bootimage --target x86_64-unknown-none.json

Best regards!

phil-opp, 2020-01-07:
I tried to reproduce the error, but a test project compiles successfully with bootloader 0.8.3 and your target file.

The error you're seeing is emitted here:

https://github.com/rust-osdev/bootloader/blob/90f5b8910d146d6d489b70a6341d778253663cfa/build.rs#L3-L7

Are you sure that you have the `arch` key in your target JSON set to "x86_64` and not "x86"?

bjorn3, 2020-01-07:
I think @0x00AE6150 is using a 32bit system. A build script always gets build for the native architecture, not the target architecture.

phil-opp, 2020-01-07:
Good catch! Moving these checks to the lib.rs/main.rs should solve the problem, right?

bjorn3, 2020-01-07:
Yes

0x00AE6150, 2020-01-07:
I am not sure why but my windows machine has 32 bit rust toolchain installed on 64 bit system... However, during rust installation i get prompted that x86_64 tools are installed. I will try to look more into it later.

phil-opp, 2020-01-08:
@0x00AE6150 It could also be that you have e.g. an ARM CPU instead of a `x86_64` CPU. Either way, the mistake is on our side, as it should be possible to build the bootloader on all platforms independent of the CPU architecture. The fix is actually relatively simple: We just have to move the checks linked above to a different file, so that they check the target system instead of the host system. I implemented this fix in https://github.com/rust-osdev/bootloader/pull/91 and will publish a new bootloader version as soon as it's merged.

closed 2020-01-13 by phil-opp

phil-opp, 2020-01-13:
@0x00AE6150 I just released the fix in #91 as version 0.8.4. Could you try whether it fixes your issue?

0x00AE6150, 2020-01-23:
@phil-opp I am so sorry for the delay. I will verify the fix today.

0x00AE6150, 2020-01-23:
@phil-opp works like a charm. Thank you very much <3

phil-opp, 2020-01-24:
@0x00AE6150 Great! Thanks a lot for verifying!

PR #91 (2020-01-13): Move architecture checks from build script into lib.rs
The build script always gets compiled for the host platform, so it only checked that the _host_ was `x86_64`. With the check in lib.rs, we correctly verify that the target is `x86_64`.

Fixes #90