rust-osdev/bootloader #93: Building error when kernel filename containing special characters

midwinter1993, 2020-02-07:
Hi all,

A building error occurs when renaming the kernel binary and building

```bash
cp example-kernel/target/x86_64-example-kernel/debug/example-kernel example-kernel/target/x86_64-example-kernel/debug/example-kernel.elf

KERNEL=example-kernel/target/x86_64-example-kernel/debug/example-kernel.elf KERNEL_MANIFEST=example-kernel/Cargo.toml cargo xbuild --release --features binary
```

Error: `rust-lld: error: undefined symbol: _kernel_start_addr`.

It seems that `build.rs` does not redefine the symbols correctly, i.e., `_binary_objfile_*` set by `objcopy`, when objfile name contains the special character `.`.

phil-opp, 2020-02-07:
Thanks for reporting this!

I created a fix in #94. Could you verify if it works for you?

midwinter1993, 2020-02-07:
Hi @phil-opp,

Great! It works for me now.

phil-opp, 2020-02-08:
Thanks for verifying!

PR #94 (2020-02-08): Objcopy replaces `.` chars with `_` chars
Adjust `stripped_kernel_file_name_replaced` for this.

Fixes #93
closed 2020-02-08 by phil-opp