rust-osdev/bootloader #95: Enable tls sections

vinaychandra, 2020-02-12:
`tdata` and `tbss` section stores thread local storage for the system. In kernel, we will need to initialize this section per-cpu or per-thread depending on usages. Rust provides a `#[thread_local]` attribute which puts the data in the above sections.

Please enable that section in bootloader and pass its location to the kernel so that it can map the TLS data as needed.

phil-opp, 2020-02-12:
Could you clarify what you mean with "enable that section"? Do you mean to just load and map it into the virtual memory or are there any additional required steps?

vinaychandra, 2020-02-12:
Just mapping into memory and then providing location is enough.
[This](https://gitlab.redox-os.org/redox-os/kernel/blob/master/src/arch/x86_64/paging/mod.rs#L202) is an example of how it could be loaded later in kernel

PR #96 (2020-02-25): Add basic support for ELF thread local storage segments
This reports the thread local storage segment of the kernel (if present) in the boot information. This should make it possible to use the unstable [`#[thread_local]` attribute](https://github.com/rust-lang/rust/issues/29594). Since the TLS template overlaps with parts of a LOAD segment, no additional mapping needs to be done.

Note that the TLS template should not be used directly. Instead, the kernel should create a copy the TLS template for each thread and then set up the `fs` register accordingly. See https://uclibc.org/docs/tls.pdf fo [... cut]
closed 2020-02-25 by phil-opp