rust-osdev/bootloader #98: BootInfo address appears to be erroneously set to 0x10000000000

zesterer, 2020-03-03:
Causing a page fault upon attempted access, of course.

phil-opp, 2020-03-04:
@zesterer Thanks for reporting! Do you have your code online somewhere so that I can try to reproduce it?

zesterer, 2020-03-06:
You can see this happening with https://github.com/tupai-os/deus.

Relevant code is here: https://github.com/tupai-os/deus/blob/master/src/arch/x86_64/mod.rs#L12

and here: https://github.com/tupai-os/deus

phil-opp, 2020-03-06:
Thanks! I just tried it and it seems like the boot info is really at this address. Accessing it works without problems for me. Try inserting `logln!("BootInfo: {:#?}", info);` at the beginning of your `kernel_start` function.

zesterer, 2020-03-06:
How bizarre. This definitely wasn't working before. Sounds like I've got a spurious issue to fix! Thanks for the rapidly response.

closed 2020-03-06 by zesterer

phil-opp, 2020-03-06:
No problem! Let me know if you encounter this issue again.