rust-osdev/bootloader #103: Int 13h extensions not supported when running produced .bin on virtualbox

fddnn, 2020-03-13:
[image]
?

bjorn3, 2020-03-13:
Just a wild guess: Is the disk attached as IDE?

fddnn, 2020-03-13:
no floppy controller would that be the cause of this?

RKennedy9064, 2020-03-14:
I found some information regarding int 13 extension issues [here](https://stackoverflow.com/questions/56764643/bochs-does-not-support-ah-42h-on-int-13h-on-disk-boot-option). According to that post:

`
Most real computers and Bochs support "int 0x13 extensions" for most/larger devices (hard drives, CD/DVD, etc), but don't support it for floppy disks.
`

So it might be likely that being attached as a floppy disk is what's causing the error.

archisgore, 2020-09-30:
I got the exact same with a floppy with VMWare. When I wrote it to an HDD, the boot worked.

fddnn, 2020-10-01:
Ok thanks that fixed it for me.

closed 2020-10-01 by fddnn