rust-osdev/bootloader #106: Running under HAXM?

ethindp, 2020-04-22:
I want to run my kernel under HAXM but whenever I do that Qemu immediately gives me two VCPU shutdown requests (that's using `-accel hax`). Is there a way we can add HAXM support or is this a bug in Qemu itself?

phil-opp, 2020-04-23:
I never worked with HAXM, so I don't know what's the issue of this. It _could_ be that we're doing something wrong in the bootloader, but it's difficult to say without any debugging information. Is is possible to get some kind of debug output with HAXM so that we know at which operation the failure occurs?

ethindp, 2020-05-13:
If there is, I have no idea how. I've tried to research this but have not met any success.

closed 2021-04-19 by ethindp