rust-osdev/bootloader #107: Deprecation: `asm!` > `llvm_asm!`

realKennyStrawn93, 2020-04-25:
Should be a pretty simple problem to fix, but as of 24 April 2020, use of the `asm!` macro is triggering multiple deprecation warnings:

    warning: use of deprecated item 'asm': the syntax of asm! will change soon, use llvm_asm! to avoid breakage
      --> src/main.rs:46:5
       |
    46 |     asm!("call $1; ${:private}.spin.${:uid}: jmp ${:private}.spin.${:uid}" ::
       |     ^^^ help: replace the use of the deprecated item: `llvm_asm`
       |
       = note: `#[warn(deprecated)]` on by default
    
    warning: use of deprecated item 'asm': the syntax of asm! will change soon, use llvm_asm! to avoid breakage
      --> src/main.rs:92:5
       |
    92 |     asm!("mov bx, 0x0
       |     ^^^ help: replace the use of the deprecated item: `llvm_asm`

Should be an easy fix if you ask me ― since there's probably a lot more instances than those two, is this being worked on already?

realKennyStrawn93, 2020-04-25:
#108

realKennyStrawn93, 2020-04-25:
Pull Request #108 includes a fix for this.

closed 2020-04-26 by phil-opp

SoulDrivenDev, 2021-09-19:
Hmm... Current nightly is giving me the opposite issue right now:

```
warning: use of deprecated macro `llvm_asm`: will be removed from the compiler, use asm! instead
  --> src/main.rs:45:5
   |
45 |     llvm_asm!("call $1; ${:private}.spin.${:uid}: jmp ${:private}.spin.${:uid}" ::
   |     ^^^^^^^^
   |
   = note: `#[warn(deprecated)]` on by default

warning: use of deprecated macro `llvm_asm`: will be removed from the compiler, use asm! instead
  --> src/main.rs:91:5
   |
91 |     llvm_asm!("mov bx, 0x0
   |     ^^^^^^^^

warning: `bootloader` (bin "bootloader") generated 2 warnings
```

Looks like `x86_64` updated to use `asm!` as well: https://github.com/rust-osdev/x86_64/issues/164

My understanding is that `asm!` was renamed to `llvm_asm!` as an intermediate step to keep it as a legacy API, but there is a new `asm!` macro that should be used.

Alonharmon1, 2021-09-28:
> Hmm... Current nightly is giving me the opposite issue right now:
> 
> ```
> warning: use of deprecated macro `llvm_asm`: will be removed from the compiler, use asm! instead
>   --> src/main.rs:45:5
>    |
> 45 |     llvm_asm!("call $1; ${:private}.spin.${:uid}: jmp ${:private}.spin.${:uid}" ::
>    |     ^^^^^^^^
>    |
>    = note: `#[warn(deprecated)]` on by default
> 
> warning: use of deprecated macro `llvm_asm`: will be removed from the compiler, use asm! instead
>   --> src/main.rs:91:5
>    |
> 91 |     llvm_asm!("mov bx, 0x0
>    |     ^^^^^^^^
> 
> warning: `bootloader` (bin "bootloader") generated 2 warnings
> ```
> 
> Looks like `x86_64` updated to use `asm!` as well: [rust-osdev/x86_64#164](https://github.com/rust-osdev/x86_64/issues/164)
> 
> My understanding is that `asm!` was renamed to `llvm_asm!` as an intermediate step to keep it as a legacy API, but there is a new `asm!` macro that should be used.

so this means that even if it will be fixed in a new version and it will be published it won't be useable in blog_os because versions 0.10.0 and newer aren't working for blog_os?

PR #108 (2020-04-26): Use `llvm_asm!` instead of deprecated `asm!`
Fixes #107

This potentially **breaks users on older nightlies**.