rust-osdev/bootloader #119: Map critical memory regions before loading kernel if detected

ethindp, 2020-06-03:
I'm wondering if its possible to explicitly map memory regions in the bootloader? For example, could we detect to see if the APIC is available, then map its entire memory region into the page table before loading the kernel, and then let the kernel configure the IDT? We can even pass the address of the APIC region (its default is FE0000, if I'm not mistaken, but this can change).

phil-opp, 2020-06-04:
What would be the advantage over doing this in the kernel? We try to keep the bootloader as small as possible and do only the work that would be impossible or much more difficult to do after the load. Detecting and mapping the APIC sounds like something that can be done in the kernel, or am I missing something?

ethindp, 2020-06-04:
I've tried to do that myself in my own kernel. For some reason its page-faulting through when trying to read memory address FEEB0h (that might just be a bug I can't seem to find though). Strangely enough, it happens immediately after I enable interrupts (by that point I've configured the PIC completely). According to OSDev I'm supposed to initialize the PIC whether an aPIC is present or not. My code is available at https://github.com/ethindp/kernel/blob/master/src/interrupts.rs.
I know I do some of the initialization sequences twice (or thrice) and I could probably trim this down by quite a lot, but its a (supposedly) good working implementation to start with. But this page faults when it hits line 115. According to GDB, this page faults as soon as it tries writing "INTR: Stage 1 initialization complete" to the serial port. Though this would normally suggest a bug in the uart_16550 crate, this doesn't make sense because that works until it reaches this point and then it dies. So that's why I suggested this. (The title is a bit misleading, sorry about that.) I'm not really sure how to fix this page fault; it doesn't seem to (necessarily) be a problem with my code -- not that I can find at least.

phil-opp, 2020-06-05:
Unfortunately I'm pretty busy right now, so that I don't have the time to help you debugging this.

I still see no real advantage of including the APIC initialization in the bootloader instead of doing it in the kernel, apart from convenience.

ethindp, 2021-03-24:
This issue is quite old and I solved it a while ago. Closing.

closed 2021-03-24 by ethindp