rust-osdev/bootloader #121: Booting from USB gets stuck with "Booting (first stage)..."

CaptainOachkatzl, 2020-06-12:
[image]

i get stuck with this screen on Oracle Virtual box, and 2 hardware devices, if i use a USB stick with the image.
it works fine if i use qemu-system-x86_64 to load the binary.
it works fine if i use qemu-system-x86_64 to boot from a USB stick with the image.

i am using this source code to build the binary (pinned to the commit i was using) https://github.com/CaptainOachkatzl/RustOS/tree/de6aaee00cc11f8274ca9b003730c7a10dd11210

in the zip are the binary and the virtualbox USB drive image i was using
[bootloader_stuck.zip](https://github.com/rust-osdev/bootloader/files/4772886/bootloader_stuck.zip)

------------------------------------------------------------------
this was the information part. i also want to say what you are doing is amazing and it helped me out a lot so thank you for that!

CaptainOachkatzl, 2020-06-13:
i checked out my own version of the bootloader and added some debugging outputs.

until this point the output works fine:

file "stage_1.s"
jump_to_second_stage:
    lea eax, [stage_2]
    jmp eax

which should afterwards call this:

file "stage_2.s"
stage_2:
    lea si, [second_stage_start_str]
    call real_mode_println

as this a print call that worked fine until this point it is hard to believe it would suddenly fail.
my guess is, that the 2nd stage either gets incorrectly loaded or not loaded at all and thus makes the jmp/program fail.

-----------------------------------------

for completeness the lines i added and called to get some debug output
protected_mode_print:
    mov bx, 0x4f4B         # white 'K' on red background
    mov eax, 0xb8f00       # note 32 bit offset
    mov word ptr ds:[eax], bx
    ret

its a copy paste of the smiley that gets printed in the program.
i moved the call to this around to confirm which parts of the program are reached by the processor.

bjorn3, 2020-06-13:
According to https://www.howtogeek.com/187721/how-to-boot-from-a-usb-drive-in-virtualbox/ the usb device has to be the first hard disk.

CaptainOachkatzl, 2020-06-13:
hi @bjorn3 , the usb device is not only the first but also the only hard disk for my virtual machine.
also it is booting from the drive, as the message "Booting (first stage)..." is part of the bootloader.
additional proof would be that it has the same behavior on 2 other (hardware) machines.

CaptainOachkatzl, 2020-06-13:
**additional findings**

in qemu and vbox:
disk address package is the same

dap: 0x0010
dap_blocks: 0x006B ( =107 blocks)
dap_buffer_addr: 0x7E00
dap_buffer_seg: 0x0000
dap_start_lba: 0x0000 0000 0000 0001

all of the values being the expected values

in qemu and vbox:
calling INT 0x13 with AH = 0x42 does NOT set the carry bit and also DOES set AH to 0x0 which indicates the reading of the disk was SUCCESSFUL.

in vbox only:
calling INT 0x10 does NOT print anymore after calling INT 0x13.
printing directly to the VGA buffer at 0xB8000 still WORKS.

CaptainOachkatzl, 2020-06-13:
tried another USB stick - same issue

CaptainOachkatzl, 2020-06-20:
the issue was that i wrote the image to USB with the command
`dd if=program.bin of=/dev/sdb1 && sync` 
because "sdb1" was showing up when i got the available devices with "df".

changing it to 
`dd if=program.bin of=/dev/sdb && sync`
solved the issue!

closed 2020-06-20 by CaptainOachkatzl

phil-opp, 2020-06-21:
Thanks for the update! Good to hear that you were able to solve your issue.