rust-osdev/bootloader #131: [BUG] PhysFrames mapped multiple times?

pitust, 2020-11-01:
Yeah, stuff gets mapped multiple times. Here is my log output [from an isa-debugcon]:
```
; STUFF HERE
MAP @ PhysFrame[4KiB](0x5f8000) to Page[4KiB](0xf001f7000)
Now mapping Seg
MAP @ PhysFrame[4KiB](0x5f8000) to Page[4KiB](0xf001f7000)
panicked at 'kernel mapping failed: Mapping(PageAlreadyMapped(PhysFrame[4KiB](0x5f8000)))', src/main.rs:268:6
```
And it just hangs. So... It seems that sections do NOT have to be 4k-aligned in ELFs. The culprit in my case was `.got`. Forcing it  to stay aligned fixed my problem. But, this should just work and not require custom linker scripts.

pitust, 2020-11-01:
I have some extra logging output from the bootloader

phil-opp, 2020-11-02:
Are you using any linker arguments? For me, the default linker script without any arguments seems to always produce aligned segments.

Either way, I agree that it would be nice if the bootloader could work with non-aligned segments.

pitust, 2020-11-06:
Indeed, my target triple is:
```json
{
    "llvm-target": "x86_64-unknown-none",
    "data-layout": "e-m:e-i64:64-f80:128-n8:16:32:64-S128",
    "arch": "x86_64",
    "target-endian": "little",
    "target-pointer-width": "64",
    "target-c-int-width": "32",
    "os": "none",
    "executables": true,
    "linker-flavor": "ld.lld",
    "linker": "rust-lld",
    "disable-redzone": true,
    "features": "-mmx,+sse",
    "panic-strategy": "unwind",
    "pre-link-args": {
        "ld.lld": [
            "/opt/cross/lib/gcc/x86_64-elf/10.2.0/libgcc.a", "--allow-multiple-definition",
        ]
    }
}
```
I need libgcc for _Unwind_* functions (i use that for stack traces). I assume that it is due to libgcc.a having a .got section which isn't 4k-aligned.

pitust, 2020-11-06:
anyway i switched to multiboot2 and first edition of the guide, because i want support for modern architectures.

pitust, 2020-11-06:
Wait, no, i got wrong args, i do indeed have a linker script, the one found on osdev.org (https://wiki.osdev.org/Bare_Bones)

phil-opp, 2020-11-07:
> i do indeed have a linker script, the one found on osdev.org (https://wiki.osdev.org/Bare_Bones)

Ah, that explains things. I think most default linker scripts align all sections properly, but a custom linker script might not. The linker script you linked does not mention any `.got` section, so the linker just places it after the last section, without any special alignment.

While I still agree that it would be nice if the `bootloader` crate supported non-aligned sections, section alignment is still recommended because otherwise you might not be able to set the correct write/executable permissions for these pages. For example, if `.text` shares a page with `.data`, that page must be mapped as both writable and executable, which is generally discouraged since attackers can easily run their injected code this way.

> anyway i switched to multiboot2 and first edition of the guide, because i want support for modern architectures.

Could you clarify what you mean with "modern architectures"? Support for UEFI booting or something else? I'm currently working on a new version of the `bootloader` with UEFI support, so it would be useful to know if something else is missing too.

pitust, 2020-11-08:
UEFI and coreboot. I know CSM is a thing, but i wouldn't rely on that. (there are some laptops that use coreboot out of the box, most notably pixelbooks and everything with System76 Open Firmware)