rust-osdev/bootloader #132: Kernel map address?

Luis-Hebendanz, 2020-11-09:
Hi there,

I wanted to find out the kernel mapping address and in the code it looks like it gets mapped at 0x400000 (4MB)
however when I debug it with gdb or print the kernel_main method I see that the base address is instead 0x200000.
Am I missing something?

phil-opp, 2020-11-11:
We currently load the kernel to the _physical_ address 4MB: 

https://github.com/rust-osdev/bootloader/blob/0898aa93337dc99db853260d13919767706966a0/src/stage_2.s#L49-L50

However, the virtual address that the kernel gets mapped to is taken directly from the ELF executable. The default linker script uses 2MB as the base address, so that's where the kernel ends up.

Luis-Hebendanz, 2020-11-11:
Ahh I see that makes sense. Thanks

closed 2020-11-11 by Luis-Hebendanz