rust-osdev/bootloader #133: Need help for documentation

Luis-Hebendanz, 2020-12-06:
Hiho,

I am trying to document the build system you have created (props by the way it's amazing). 
It's located here: https://github.com/Luis-Hebendanz/bootloader/tree/smp_support#build-chain
I understand pretty much everything except the command:
```bash
llvm-objcopy "-I" "binary" "-O" "elf64-x86-64" "--binary-architecture=i386:x86-64" "--rename-section" ".data=.kernel" "--redefine-sym" "_binary_kernel_stripped_svm_kernel_start=_kernel_start_addr" "--redefine-sym" "_binary_kernel_stripped_svm_kernel_end=_kernel_end_addr" "--redefine-sym" "_binary_kernel_stripped_svm_kernel_size=_kernel_size" "target/x86_64-bootloader/debug/build/bootloader-c8df27c930d8f65a/out/kernel_stripped-svm_kernel" "target/x86_64-bootloader/debug/build/bootloader-c8df27c930d8f65a/out/kernel_bin-svm_kernel.o"
```
I get the redefinition of the .data section to .kernel for the sake of the linker script. However I do not understand where the symbols `_binary_kernel_stripped_svm_kernel_size` come from.  When I execute `readelf -a _binary_kernel_stripped_svm_kernel_size | grep _binary_kernel_stripped_svm_kernel_size` I get no results. Where do these syms come from?

bjorn3, 2020-12-06:
From `llvm-objcopy --help`: `--redefine-sym=old=new  Change the name of a symbol old to new`. `_binary_kernel_stripped_svm_kernel_start` is the old symbol name. `_kernel_start_addr` is what ends up in the final binary. I can't find where `_binary_kernel_stripped_svm_kernel_start` comes from though.

Luis-Hebendanz, 2020-12-06:
> From `llvm-objcopy --help`: `--redefine-sym=old=new Change the name of a symbol old to new`. `_binary_kernel_stripped_svm_kernel_start` is the old symbol name. `_kernel_start_addr` is what ends up in the final binary. I can't find where `_binary_kernel_stripped_svm_kernel_start` comes from though.

Well that was my question ;)

Luis-Hebendanz, 2020-12-09:
@phil-opp Any hints? :-)

phil-opp, 2020-12-10:
@Luis-Hebendanz Thanks a lot for your work on this! It would be really great to finally have some documentation on this. Unfortunately, I'm relatively busy at the moment, so I don't have the time to take a closer look at your writeup right now, but on first glance it looks pretty good.

Regarding the `objcopy` command: It is defined in the build script here:

https://github.com/rust-osdev/bootloader/blob/0898aa93337dc99db853260d13919767706966a0/build.rs#L192-L203

The `stripped_kernel_file_name_replaced` variable here is the binary name of the stripped kernel (stripping is part of the build process), with `-` and `.` replaced by `_` (as `objcopy` performs these replacements automatically). The `_binary_{}_start`, `_binary_{}_end`, and `_binary_{}_size` symbols are automatically created by `objcopy` when invoking it with `"-I" "binary" "-O" "elf64-x86-64"`. They mark the start address, end address, and size of the embedded binary.

Hope this helps! Let me know if you have further questions.

Luis-Hebendanz, 2020-12-10:
I see that makes sense! Now I have everything I need, thanks. I will create a pull request later on,
 if you don't have time reviewing it, don't stress about it, it can wait :-)

closed 2020-12-10 by Luis-Hebendanz