rust-osdev/bootloader #136: The symbol _kernel_start_addr is undefined

ivkin25, 2021-02-15:
I am trying to use the symbol `_kernel_start_addr` along with `_kernel_end_addr` and `_kernel_size` in my OS that's loaded by the bootloader, but the linker says that these symbols are undefined:

main.rs:
```rust
...

extern "C" {
    static _kernel_start_addr: usize;
    static _kernel_end_addr: usize;
    static _kernel_size: usize;
}

entry_point!(kernel_main);

pub fn kernel_main(boot_info: &'static BootInfo) -> ! {
    ...
    
    unsafe {
        println!("_kernel_start_addr: {}", _kernel_start_addr);
        println!("_kernel_end_addr: {}", _kernel_end_addr);
        println!("_kernel_size: {}", _kernel_size);
    }

   ...
}
```

```
...
= note: rust-lld: error: undefined symbol: _kernel_start_addr
          >>> referenced by main.rs:0 (<path>\src\main.rs:0)
          >>>               <path>\target\x86_64-rustos-target\debug\deps\rustos-1269839a7727b609.3aqbiipwufbmgq53.rcgu.o:(rustos::kernel_main::h036cbed503907ed8)

          rust-lld: error: undefined symbol: _kernel_end_addr
          >>> referenced by main.rs:0 (<path>\src\main.rs:0)
          >>>               <path>\target\x86_64-rustos-target\debug\deps\rustos-1269839a7727b609.3aqbiipwufbmgq53.rcgu.o:(rustos::kernel_main::h036cbed503907ed8)

          rust-lld: error: undefined symbol: _kernel_size
          >>> referenced by main.rs:0 (<path>\src\main.rs:0)
          >>>               <path>\target\x86_64-rustos-target\debug\deps\rustos-1269839a7727b609.3aqbiipwufbmgq53.rcgu.o:(rustos::kernel_main::h036cbed503907ed8)
```

And the linker is not wrong. I run `nm` and I really don't see that these symbols are stored in the kernel's symbol table. Am I missing something?

Thanks!

phil-opp, 2021-02-15:
If you need such symbols, you have to define them yourself in a custom linker script. The `_kernel_start_addr` etc addresses are used to describe the beginning/end of the kernel binary file within the bootloader executable.

The linker is in fact invoked twice: First, you link your kernel as an ELF executable. Second, the bootloader is built and linked, storing your kernel executable in a `static` variable. The kernel executable is only stripped (debug symbols removed), but not modified otherwise. Thus, the kernel linking is completely independent of the bootloader linking.

ivkin25, 2021-02-15:
Got it. Is there a way that I can make the bootloader run my custom linker script, in addition to its script?

phil-opp, 2021-02-15:
Why do you want to run a custom linker script for the bootloader instead of for the kernel?

ivkin25, 2021-02-15:
Sorry, mistyped the wrong term.
I actually can sort it out by modifying the custom target file, and specifying the linker script. Thanks!

closed 2021-02-15 by ivkin25