rust-osdev/bootloader #141: Access RSDP from BootInfo

Andy-Python-Programmer, 2021-03-22:
Is there a way to access the RSDP from boot info? 

<https://wiki.osdev.org/RSDP>

phil-opp, 2021-03-22:
Not yet, but the upcoming version in #130 will include the RSDP in the boot info. You can try to use https://docs.rs/acpi/2.3.0/acpi/struct.AcpiTables.html#method.search_for_rsdp_bios directly until then.

Andy-Python-Programmer, 2021-03-24:
Is there any example to help integrate it with boot loader. As it requires a handler. Do I have to make my own one it’s not documented @phil-opp

phil-opp, 2021-03-24:
The function works by scanning certain physical memory regions for the RSDP signature (see https://wiki.osdev.org/RSDP#Detecting_the_RSDP ). For this is needs access to physical memory. This is what the handler is for.

You can create your own handler by implementing the [`AcpiHandler`](https://docs.rs/acpi/2.3.0/acpi/trait.AcpiHandler.html) trait. For this you need to provide methods to map and unmap physical memory pages. If you're using the `map-physical-memory` feature of the bootloader, you don't need to do any actual mapping in it as you can just add the offset. 

For an example implementation, this is how we implemented it in the upcoming new version of the `bootloader` crate:

https://github.com/rust-osdev/bootloader/blob/873351c575bdefd1c6c78b27de2bc0494698c0d5/src/bin/bios.rs#L193-L225

Instead of an identity mapping, you probably have the physical memory mapped at an offset, which you need to add to the address when filling the `virtual_start` field of the `PhysicalMapping`.

Hope this helps!

Andy-Python-Programmer, 2021-03-24:
It does. Thanks!

Andy-Python-Programmer, 2021-04-08:
Closing as version `0.10.0` is released!

closed 2021-04-08 by Andy-Python-Programmer