rust-osdev/bootloader #142: Bootloader doesn't compile

JulianKnodt, 2021-03-26:
Hi,

Thank you for the bootloader, I've been following Phil Opperman's Blog and encountered an issue after updating to the latest nightly where this crate no longer compiles with a linker error:

```
    Finished dev [unoptimized + debuginfo] target(s) in 0.05s
     Running `bootimage runner target/target/debug/xokernel`
warning: panic message is not a string literal
   --> build.rs:128:9
    |
128 |         format!("KERNEL does not exist: {}", kernel.display())
    |         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    |
    = note: `#[warn(non_fmt_panic)]` on by default
    = note: this is no longer accepted in Rust 2021
    = note: the panic!() macro supports formatting, so there's no need for the format!() macro here
help: remove the `format!(..)` macro call
    |
128 |         "KERNEL does not exist: {}", kernel.display()
    |        --                                           --

warning: 1 warning emitted

   Compiling bootloader v0.9.12 (/Users/julianknodt/.cargo/registry/src/github.com-1ecc6299db9ec823/bootloader-0.9.12)
error: linking with `rust-lld` failed: exit code: 1
  |
  = note: "rust-lld" "-flavor" "gnu" "--script=linker.ld" "--eh-frame-hdr" "-L" "/Users/julianknodt/.rustup/toolchains/nightly-x86_64-apple-darwin/lib/rustlib/x86_64-bootloader/lib" "/Users/julianknodt/Desktop/programming/projects/rustjects/xokernel/target/bootimage/bootloader/x86_64-bootloader/release/deps/bootloader-57fef89eac316b5b.bootloader.dcl21vxn-cgu.0.rcgu.o" "/Users/julianknodt/Desktop/programming/projects/rustjects/xokernel/target/bootimage/bootloader/x86_64-bootloader/release/deps/bootloader-57fef89eac316b5b.bootloader.dcl21vxn-cgu.1.rcgu.o" "/Users/julianknodt/Desktop/programming/projects/rustjects/xokernel/target/bootimage/bootloader/x86_64-bootloader/release/deps/bootloader-57fef89eac316b5b.bootloader.dcl21vxn-cgu.10.rcgu.o" "/Users/julianknodt/Desktop/programming/projects/rustjects/xokernel/target/bootimage/bootloader/x86_64-bootloader/release/deps/bootloader-57fef89eac316b5b.bootloader.dcl21vxn-cgu.11.rcgu.o" "/Users/julianknodt/Desktop/programming/projects/rustjects/xokernel/target/bootimage/bootloader/x86_64-bootloader/release/deps/bootloader-57fef89eac316b5b.bootloader.dcl21vxn-cgu.12.rcgu.o" "/Users/julianknodt/Desktop/programming/projects/rustjects/xokernel/target/bootimage/bootloader/x86_64-bootloader/release/deps/bootloader-57fef89eac316b5b.bootloader.dcl21vxn-cgu.13.rcgu.o" "/Users/julianknodt/Desktop/programming/projects/rustjects/xokernel/target/bootimage/bootloader/x86_64-bootloader/release/deps/bootloader-57fef89eac316b5b.bootloader.dcl21vxn-cgu.14.rcgu.o" "/Users/julianknodt/Desktop/programming/projects/rustjects/xokernel/target/bootimage/bootloader/x86_64-bootloader/release/deps/bootloader-57fef89eac316b5b.bootloader.dcl21vxn-cgu.15.rcgu.o" "/Users/julianknodt/Desktop/programming/projects/rustjects/xokernel/target/bootimage/bootloader/x86_64-bootloader/release/deps/bootloader-57fef89eac316b5b.bootloader.dcl21vxn-cgu.2.rcgu.o" "/Users/julianknodt/Desktop/programming/projects/rustjects/xokernel/target/bootimage/bootloader/x86_64-bootloader/release/deps/bootloader-57fef89eac316b5b.bootloader.dcl21vxn-cgu.3.rcgu.o" "/Users/julianknodt/Desktop/programming/projects/rustjects/xokernel/target/bootimage/bootloader/x86_64-bootloader/release/deps/bootloader-57fef89eac316b5b.bootloader.dcl21vxn-cgu.4.rcgu.o" "/Users/julianknodt/Desktop/programming/projects/rustjects/xokernel/target/bootimage/bootloader/x86_64-bootloader/release/deps/bootloader-57fef89eac316b5b.bootloader.dcl21vxn-cgu.5.rcgu.o" "/Users/julianknodt/Desktop/programming/projects/rustjects/xokernel/target/bootimage/bootloader/x86_64-bootloader/release/deps/bootloader-57fef89eac316b5b.bootloader.dcl21vxn-cgu.6.rcgu.o" "/Users/julianknodt/Desktop/programming/projects/rustjects/xokernel/target/bootimage/bootloader/x86_64-bootloader/release/deps/bootloader-57fef89eac316b5b.bootloader.dcl21vxn-cgu.7.rcgu.o" "/Users/julianknodt/Desktop/programming/projects/rustjects/xokernel/target/bootimage/bootloader/x86_64-bootloader/release/deps/bootloader-57fef89eac316b5b.bootloader.dcl21vxn-cgu.8.rcgu.o" "/Users/julianknodt/Desktop/programming/projects/rustjects/xokernel/target/bootimage/bootloader/x86_64-bootloader/release/deps/bootloader-57fef89eac316b5b.bootloader.dcl21vxn-cgu.9.rcgu.o" "-o" "/Users/julianknodt/Desktop/programming/projects/rustjects/xokernel/target/bootimage/bootloader/x86_64-bootloader/release/deps/bootloader-57fef89eac316b5b" "--gc-sections" "-L" "/Users/julianknodt/Desktop/programming/projects/rustjects/xokernel/target/bootimage/bootloader/x86_64-bootloader/release/deps" "-L" "/Users/julianknodt/Desktop/programming/projects/rustjects/xokernel/target/bootimage/bootloader/release/deps" "-L" "/Users/julianknodt/Desktop/programming/projects/rustjects/xokernel/target/bootimage/bootloader/x86_64-bootloader/release/build/bootloader-5d9ac03bd71d2b04/out" "-L" "/Users/julianknodt/.rustup/toolchains/nightly-x86_64-apple-darwin/lib/rustlib/x86_64-bootloader/lib" "-Bstatic" "/Users/julianknodt/Desktop/programming/projects/rustjects/xokernel/target/bootimage/bootloader/x86_64-bootloader/release/deps/libxmas_elf-2a81361439a261d7.rlib" "/Users/julianknodt/Desktop/programming/projects/rustjects/xokernel/target/bootimage/bootloader/x86_64-bootloader/release/deps/libzero-96434f58cbba2535.rlib" "/Users/julianknodt/Desktop/programming/projects/rustjects/xokernel/target/bootimage/bootloader/x86_64-bootloader/release/deps/libx86_64-9328d043ac4ae391.rlib" "/Users/julianknodt/Desktop/programming/projects/rustjects/xokernel/target/bootimage/bootloader/x86_64-bootloader/release/deps/libbitflags-f7fa929c822f7eb3.rlib" "/Users/julianknodt/Desktop/programming/projects/rustjects/xokernel/target/bootimage/bootloader/x86_64-bootloader/release/deps/libbit_field-cb737cce5c9f5c0c.rlib" "/Users/julianknodt/Desktop/programming/projects/rustjects/xokernel/target/bootimage/bootloader/x86_64-bootloader/release/deps/libusize_conversions-5d40b777e96d358e.rlib" "/Users/julianknodt/Desktop/programming/projects/rustjects/xokernel/target/bootimage/bootloader/x86_64-bootloader/release/deps/libfixedvec-63a3cceec10f2498.rlib" "/Users/julianknodt/Desktop/programming/projects/rustjects/xokernel/target/bootimage/bootloader/x86_64-bootloader/release/deps/libbootloader-25b9e68f64f23452.rlib" "/Users/julianknodt/Desktop/programming/projects/rustjects/xokernel/target/bootimage/bootloader/x86_64-bootloader/release/deps/librlibc-2378eec41f86e4b9.rlib" "/Users/julianknodt/Desktop/programming/projects/rustjects/xokernel/target/bootimage/bootloader/x86_64-bootloader/release/deps/librustc_std_workspace_core-077cd7bb4af44f0e.rlib" "/Users/julianknodt/Desktop/programming/projects/rustjects/xokernel/target/bootimage/bootloader/x86_64-bootloader/release/deps/libcore-e07b7db51e47458f.rlib" "/Users/julianknodt/Desktop/programming/projects/rustjects/xokernel/target/bootimage/bootloader/x86_64-bootloader/release/deps/libcompiler_builtins-8d89cd04c4385fe3.rlib" "-Bdynamic"
  = note: rust-lld: error: bootloader.dcl21vxn-cgu.1:(.boot-first-stage+0x81): relocation R_X86_64_16 out of range: 87040 is not in [-32768, 65535]; references _rest_of_bootloader_end_addr
          rust-lld: error: bootloader.dcl21vxn-cgu.1:(.boot+0x6A): relocation R_X86_64_16 out of range: 87040 is not in [-32768, 65535]; references _kernel_start_addr
          >>> defined in /Users/julianknodt/Desktop/programming/projects/rustjects/xokernel/target/bootimage/bootloader/x86_64-bootloader/release/deps/libbootloader-25b9e68f64f23452.rlib(kernel_bin-xokernel.o)
          
          rust-lld: error: bootloader.dcl21vxn-cgu.1:(.boot+0x85): relocation R_X86_64_16 out of range: 165384 is not in [-32768, 65535]; references _kernel_size
          >>> defined in /Users/julianknodt/Desktop/programming/projects/rustjects/xokernel/target/bootimage/bootloader/x86_64-bootloader/release/deps/libbootloader-25b9e68f64f23452.rlib(kernel_bin-xokernel.o)
          

error: aborting due to previous error

error: could not compile `bootloader`

To learn more, run the command again with --verbose.
Error: Bootloader build failed.
```

Is this an issue on my end/Is there any contribution I can add to easily fix this?

Thank you!

bjorn3, 2021-03-26:
You need to update the bootloader crate. `cargo update -p bootloader`.

JulianKnodt, 2021-03-26:
brilliant, that works, how would I know when I need to update it in the future?

closed 2021-03-26 by JulianKnodt