rust-osdev/bootloader #144: Bootloader not working with -device q35

Andy-Python-Programmer, 2021-03-28:
Bootloader does not work with `-device q35` argument passed to qemu to access the MCFG tables and SATA. Heres the output:

[image]

phil-opp, 2021-03-28:
Hmm, I never tried to use q35. Looks like it doesn't even detect the boot disk. Could you try whether it works with other system images? This would tell us whether the problem is in the disk image generated by this crate or whether it is in QEMU itself.

Andy-Python-Programmer, 2021-03-28:
> Hmm, I never tried to use q35. Looks like it doesn't even detect the boot disk. Could you try whether it works with other system images? This would tell us whether the problem is in the disk image generated by this crate or whether it is in QEMU itself.

I did try it with another image before I tried it with this crate and it seems to work. This crate seems to be the err. @phil-opp

phil-opp, 2021-03-28:
Thanks for the quick reply! I suspect that is because we currently only write the boot signature `0x55aa`, but not any partition table. I'll try to fix this soon, but I'm currently still busy with the rewrite for UEFI support.

phil-opp, 2023-01-02:
We now create a proper FAT partition for booting. Does this issue still occur with `v0.11`?