rust-osdev/bootloader #145: Global Page Flag & Kernel Mapping

MarcoCicognani, 2021-03-28:
Hi,
as I written in the title it would be a good idea to enable global pages and map the kernel using this flag.
Doing this any kernel mapping would be implicitly duplicated across all the address spaces.

Keep it up!
Waiting UEFI rework's release

bjorn3, 2021-03-28:
Not every kernel may want to do this. For example it could use [kernel page table isolation](https://en.wikipedia.org/wiki/Kernel_page-table_isolation) as mitigation for Meltdown. In addition the kernel can map itself using this flag.

MarcoCicognani, 2021-03-28:
Sure! The it could be a configurable feature like recursive or phisical mapping

phil-opp, 2021-04-27:
We typically only provide functionality in the bootloader that is impossible or difficult to implement in the kernel itself. For example, it is not possible to set up a physical memory mapping without access to the page tables in the first place. Setting the global bit, on the other hand, can be done relatively easily from the kernel itself by iterating over the non-empty page table entries and setting the global bit for each.

So adding support for this in the bootloader would be purely a convenience feature. It might be worth it if enough people want it, but I'm not sure if the approach is still common enough today (given Meltdown and similar vulnerabilities).