rust-osdev/bootloader #146: Docs on how to use UEFI feature

Andy-Python-Programmer, 2021-04-07:
Currently if I run my os:
```toml
[dependencies.bootloader]
version = "0.10.0"
features = ["binary", "uefi_bin"]
```

I get:
```
   Compiling bootloader v0.10.0
error: failed to run custom build command for `bootloader v0.10.0`

Caused by:
  process didn't exit successfully: `H:\Andy\Rust\aero\target\debug\build\bootloader-2cdadb50cdaed247\build-script-build` (exit code: 101)
  --- stderr
  thread 'main' panicked at 'The UEFI bootloader must be compiled for the `x86_64-unknown-uefi` target.', C:\Users\user\.cargo\registry\src\github.com-1ecc6299db9ec823\bootloader-0.10.0\build.rs:41:13
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
  ```
  
  Which is kindof obvious but until bootimage get support for it?

Andy-Python-Programmer, 2021-04-07:
Also some docs on windows user how to get OVMF.fd without WSL if thats possible :)

phil-opp, 2021-04-07:
See https://docs.rs/bootloader/0.10.1/bootloader/ for some instructions on how to create the bootable disk images. The idea is that you create your own "boot" crate instead of using `bootimage`. Unfortunately there is no example project yet, but I'm working on an step-by-step explanation [here](https://github.com/phil-opp/blog_os/blob/edition-3/blog/content/edition-3/posts/02-booting/index.md#bootable-disk-image).

Andy-Python-Programmer, 2021-04-07:
All good! I will check it out thanks!

phil-opp, 2021-04-07:
> Also some docs on windows user how to get OVMF.fd without WSL if thats possible will be great :)

Absolutely, thanks for the suggestion! I'm still figuring out the best way to do this, but it seems like downloading a precompiled version is the easiest solution. The question is where from... For now, I think you should be able to use the precompiled versions of the [pebble](https://github.com/IsaacWoods/pebble/tree/master/bundled/ovmf) project.

georglauterbach, 2022-02-03:
@phil-opp is there a flag for the bootloader to only build UEFI, or better, only the `boot-uefi-<kernel>.efi` file? I do not need the other files, and they waste time so to speak.

georglauterbach, 2022-02-03:
Thanks for your response. I found the flag I need - it was `--firmware uefi` when invoking the `builder` command to just build UEFI related files :)