rust-osdev/bootloader #149: Failing to build

Sable-20, 2021-04-09:
# **BUG**

### *error log*
```
error: failed to run custom build command for `bootloader v0.10.1 (C:\Users\Sable\.cargo\registry\src\github.com-1ecc6299db9ec823\bootloader-0.10.1)`

Caused by:
  process didn't exit successfully: `C:\Users\Sable\.cargo\registry\src\github.com-1ecc6299db9ec823\bootloader-0.10.1\target\release\build\bootloader-54e4c46467713bf6\build-script-build` (exit code: 101)
  --- stderr
  thread 'main' panicked at 'unexpected llvm-size line output', build.rs:95:43
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
warning: build failed, waiting for other jobs to finish...
error: build failed
thread 'main' panicked at 'assertion failed: cmd.status()?.success()', src\bin\builder.rs:112:9
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

this is with the full file path to my kernel.

```
cargo builder --kernel-manifest C:\Users\Sable\rust_os\Cargo.toml --kernel-binary C:\Users\Sable\rust_os\target\x86_64-rust_os\debug
```

the issue for me, and im not sure if its whats causing the error, is that while building my kernel, *in the kernels directory* it does not produce a `.bin` file in my target directory anywhere. on top of that its telling me that the llvm-size line output was unexpected. If you could follow up or maybe provide some insight id be grateful

phil-opp, 2021-04-11:
Thanks for reporting! Unfortunately the error message is not that useful because it doesn't show the actual `llvm-size` output. I just pushed 754af6ccdabf327ce455ffd32c1017292013ebac to the `main` branch to improve this. Could you try again using the `main` branch version? You can do it like this:

```toml
[dependencies]
bootloader = { git = "https://github.com/rust-osdev/bootloader.git" }
```

Sable-20, 2021-04-18:
> Thanks for reporting! Unfortunately the error message is not that useful because it doesn't show the actual `llvm-size` output. I just pushed [754af6c](https://github.com/rust-osdev/bootloader/commit/754af6ccdabf327ce455ffd32c1017292013ebac) to the `main` branch to improve this. Could you try again using the `main` branch version? You can do it like this:
> 
> ```toml
> [dependencies]
> bootloader = { git = "https://github.com/rust-osdev/bootloader.git" }
> ```

Hi! Sorry for the late reply! I will of course test it but there were other issues that I hadnt resolved. Since the updates the code on your blog no longer works and I had to make a few changes regarding struct names and referencing the variables. This was no issue of yours I merely didnt notice that there were some files I missed struct name changes in.

So it would work if I did this but thank you so much for making the effort to look into my issue even though it was just me overlooking simple things.