rust-osdev/bootloader #151: How to connect bootloader to kernel

JakubGawron23, 2021-04-19:
How to properly connect the bootloader with the kernel so that the `` entry_point! '' Macros can be used
[image]

phil-opp, 2021-04-19:
What error message are you seeing?

JakubGawron23, 2021-04-19:
My point is how to link the bootloader as a folder and not as a dependency in Cargo.toml.

phil-opp, 2021-04-19:
You mean that you cloned the bootloader locally and want to use that folder as a dependency instead of crates.io? In that case, you can specify a [path dependency](https://doc.rust-lang.org/cargo/reference/specifying-dependencies.html#specifying-path-dependencies) in your Cargo.toml.

JakubGawron23, 2021-04-19:
I added the path to Cargo.toml and ran all the commands and got such an error, how to fix it?
[image]

JakubGawron23, 2021-04-19:
However, the problem was solved :)

closed 2021-04-19 by JakubGawron23

phil-opp, 2021-04-19:
Great to hear that!