rust-osdev/bootloader #152: RSDP reporting

genderev, 2021-04-27:
```
rsdp_addr: Optional<u64>

The address of the RSDP data structure, which can be use to find the ACPI tables.

This field is None if no RSDP was found (for BIOS) or reported (for UEFI).
```
In what situation is RSDP not reported? If the RSDP is important to your kernel, is there any way to guarantee it will be `u64` and not `None`?

```
let gop = unsafe { &mut *gop.get() }; 
```

What is the purpose of the above line from `src/bin/uefi.rs`?

bjorn3, 2021-04-27:
> In what situation is RSDP not reported? If the RSDP is important to your kernel, is there any way to guarantee it will be u64 and not None?

No RSDP is reported when the firmware doesn't support ACPI for whatever reason. For example because it is too old.

> What is the purpose of the above line from `src/bin/uefi.rs`?

`gop` is of type `&UnsafeCell<GraphicsOutput>`. This line converts it to `&mut GraphicsOutput` which is unsafe as it requires that there exists no other reference to the same location of a type unless it is also an `&UnsafeCell<GraphicsOutput>`.

genderev, 2021-04-27:
Thank you for clarifying. 

Can you perform an operation such as `let gop = unsafe { &mut *gop.get() };` for other boot services or is this specific to the graphics protocol?

bjorn3, 2021-04-27:
It would work with any boot service. Just be careful to not have more than one non `&UnsafeCell<T>` reference to each boot service at any given time.

genderev, 2021-04-27:
Thanks. I hope it is okay if I ask questions later as they arise - I have read a lot about UEFI but most documentation is in C.

closed 2021-04-27 by genderev