rust-osdev/bootloader #158: Support for multiple boot protocols

Andy-Python-Programmer, 2021-05-09:
It will be great to have support for multiple boot protocols preferably stivale, stivale2 and mutliboot2. Other boot protocols also exist but these are the "cool" ones :D

<https://wiki.osdev.org/Stivale_Bare_Bones>

phil-opp, 2021-05-13:
Do you mean that we should support booting kernels using these protocols? Or that we should support to create disk images that can be booted by other bootloaders (e.g. GRUB2) via these protocols?

Andy-Python-Programmer, 2021-05-13:
Support booting the kernel with these protocols.

phil-opp, 2021-05-14:
Hmm, I'm not sure if this is worth the effort. It would require a lot of boilerplate to set up the various C-style structures that these protocols require (e.g. for the boot info). It would also make this crate much more complex because we would need to support all the machine states required by these protocols. For example, we would need to also support loading 32 bit ELF files into protected mode for multiboot2 support (we currently do everything in 64 bit).

Andy-Python-Programmer, 2021-05-14:
Ah I see