rust-osdev/bootloader #159: Higher Half?

Andy-Python-Programmer, 2021-05-13:
Will be nice to support higher half kernels.

phil-opp, 2021-05-13:
Have you tried to specify a custom virtual base address via an `--image-base <addr>` argument when building your kernel? Alternatively, you can use a linker script for this. The bootloader should be able to load the kernel into the virtual address space at exactly the specified address, which should also be possible with addresses in the higher half. I say "should" because I have not tested this yet since the UEFI rewrite. Please let me know if it doesn't work so that we can fix it!

Andy-Python-Programmer, 2021-05-14:
> Alternatively, you can use a linker script for this. 
I used the following linker script and is verified on the limine bootloader that it works fine and the higher offset value is fine.

```ld
ENTRY(_start)
OUTPUT_FORMAT(elf64-x86-64)

KERNEL_OFFSET = 0xFFFF800000100000;

SECTIONS {
    . = KERNEL_OFFSET;

    . += SIZEOF_HEADERS;
    . = ALIGN(4096);

    .text : AT(ADDR(.text) - KERNEL_OFFSET) {
        __text_start = .;
        *(.text*)
        . = ALIGN(4096);
        __text_end = .;
    }

    .rodata : AT(ADDR(.rodata) - KERNEL_OFFSET) {
        __rodata_start = .;
        *(.rodata*)
        . = ALIGN(4096);
        __rodata_end = .;
    }

    .data : AT(ADDR(.data) - KERNEL_OFFSET) {
        __data_start = .;
        *(.data*)
        . = ALIGN(4096);
        __data_end = .;
        __bss_start = .;
        *(.bss*)
        . = ALIGN(4096);
        __bss_end = .;
    }

    .tdata : AT(ADDR(.tdata) - KERNEL_OFFSET) {
        __tdata_start = .;
        *(.tdata*)
        . = ALIGN(4096);
        __tdata_end = .;
        __tbss_start = .;
        *(.tbss*)
        . += 8;
        . = ALIGN(4096);
        __tbss_end = .;
    }

    __kernel_end = .;

    /DISCARD/ : {
        *(.comment*)
        *(.eh_frame*)
        *(.gcc_except_table*)
        *(.note*)
        *(.rel.eh_frame*)
    }
}
```

[image]

In the elf loader you probablly have to subtract the KERNEL_OFFSET from the elf section's vaddr.

phil-opp, 2021-05-14:
Ah, I think I found the issue. We're reserving only two frames to identity map the context switch function, but that is not enough when the kernel lives in a different level 3 table because then we need to create new level 3/2/1 tables as well.

Regarding the linker script: You shouldn't need the `AT` hack at all (after I fixed the mentioned bug). The bootloader should be capable of loading kernels directly to their higher half addresses.

Andy-Python-Programmer, 2021-05-14:
> Ah, I think I found the issue. We're reserving only two frames to identity map the context switch function, but that is not enough when the kernel lives in a different level 3 table because then we [...]

So is it like a quick fix?

Andy-Python-Programmer, 2021-05-14:
@phil-opp I added a fix but I am still not sure as my heap does not work after this though. 

```rs
pub const HEAP_START: usize = 0x444444440000;
pub const HEAP_SIZE: usize = 100 * 1024;
```

phil-opp, 2021-05-14:
@Andy-Python-Programmer Were you able to solve the heap problem?

closed 2021-05-14 by phil-opp

Andy-Python-Programmer, 2021-05-14:
> @Andy-Python-Programmer Were you able to solve the heap problem?

Yeah thanks!

PR #161 (2021-05-14): Fix higher half kernels by identity mapping context switch fn earlier
We used to reverve two frames to allow the context switch function to identity-map itself after creating the boot info. This works for kernels that use the standard mapping since they don't need to create new level 3/2 page tables for that. For kernels that are mapped differently (e.g. to the higher half), two frames aren't enough and a frame allocation error occurs. While we could fix this by reserving a larger number of frames, this would result in lost frames for normally-mapped kernels. So we instead do the identity-mapp [... cut]