rust-osdev/bootloader #164: Pass `UnwindInfo` in `BootInfo`

Andy-Python-Programmer, 2021-05-15:
It will be nice to pass `UnwindInfo` in `BootInfo` so that we can load the kernel ELF in the kernel to do stack unwinding.

```rust
#[derive(Debug, Copy, Clone)]
#[repr(C)]
pub struct UnwindInfo {
    /// The base address of the kernel. The kernel base is required
    /// for stack unwinding during kernel panics.
    pub kernel_base: VirtAddr,
    /// The size of the kernel, required to calculate the end of the
    /// kernel base.
    pub kernel_size: usize,

    pub stack_top: VirtAddr,
    pub stack_size: usize
}
```

phil-opp, 2021-05-16:
Sounds reasonable to me! While you can already find out the kernel start address and size through a linker script, I agree that a boot info field would be easier. For stack top and size there is currently no workaround except for setting these to fixed values in the bootloader config.

ethindp, 2021-05-21:
@phil-opp Couldn't you just pass that after finding the address in the bootloader? I'm confused about the problem.

phil-opp, 2021-05-22:
There is no problem here, we just need to do it. Happy to merge a PR for this!

phil-opp, 2021-05-22:
Some adjustments to the code from the OP that I would make:

```rust
#[derive(Debug, Copy, Clone)]
#[repr(C)]
pub struct KernelInfo {
    pub kernel_base: u64
    pub kernel_size: u64,
    pub stack_top: u64,
    pub stack_size: u64,
}
```

Basically:

- Name the struct `KernelInfo` instead of `UnwindInfo` since it isn't directly related to unwinding and might have other use cases too.
- Don't use the `VirtAddr` type of the `x86_64` crate to keep the `BootInfo` stable (otherwise updating the `x86_64` version would be a breaking change).
- Don't use platform dependent size types such as `usize`. Use `u64` instead. (This might be overcautious, but it makes sure that the `BootInfo` always has the same size independent of the target platform.)