rust-osdev/bootloader #167: Bootloader v0.9.17 seems to be broken on nightly-2021-05-20

phil-opp, 2021-05-20:
I just noticed this in the automated CI runs for the `blog_os` project: https://github.com/phil-opp/blog_os/runs/2626134888?check_suite_focus=true#step:12:47

When trying to run `cargo +nightly-2021-05-20 run`, the following error occurs:

[image]

The error message is "Failed to load rest of bootloader" and happens in the early assembly stages.

I'm not sure what's going wrong yet, I will update this issue when I find the cause.

phil-opp, 2021-05-20:
Diff between the working and broken nightly: https://github.com/rust-lang/rust/compare/4e3e6db01...f94942d84

bjorn3, 2021-05-20:
`--gc-sections is now only passed if `"linker-is-gnu": true` is used inside the target spec. The blog os target spec doesn't mention it. Maybe the first stage is too big without `--gc-sections`?

phil-opp, 2021-05-20:
@bjorn3 Ah thanks, I think that's likely the reason. I'll push a fix in a moment.

phil-opp, 2021-05-20:
I opened #168 with a fix.

closed 2021-05-20 by phil-opp

phil-opp, 2021-05-20:
Fix published as v0.9.18. A `cargo update -p bootloader` should be enough to fix this issue in downstream crates.

PR #168 (2021-05-20): Fix nightly regression by manually passing `--gc-sections`
Rust used to pass `--gc-section` automatically. In https://github.com/rust-lang/rust/pull/85274, this behavior was changed to only pass that flag for targets that use a GNU linker. So we have to add it manually, otherwise the bootloader grows too large to be loaded by our assembly stage.

Fixes #167

This is a hotfix for the `v0.9.17` release and does not apply to the `main` branch (which needs some additional fixes).