rust-osdev/bootloader #172: ]Unable to get RSDP for ACPI v2 in some cases

gifnksm, 2021-05-22:
The value stored in `BootInfo::rsdp_addr` is the first one found in the RSDP of ACPI v1 and ACPI v2 ([source code]).

[source code]: https://github.com/rust-osdev/bootloader/blob/9589cff9fac705b10a659f048bf55f2a7bc160a2/src/bin/uefi.rs#L59

In my environment (QEMU), the RSDP for ACPI v1 is always stored.
Is it possible to implement a way to specify the ACPI version in the RSDP passed to the kernel?

Alternatively, there may be a way to pass the `system_table` to the kernel so that the kernel can use the UEFI runtime features.

bjorn3, 2021-05-22:
According to https://stackoverflow.com/questions/60137506/qemu-support-for-acpi-2-0 the SeaBIOS version used by qemu may be too old.

gifnksm, 2021-05-22:
I don't think it's a QEMU problem, because I was able to correctly access ACPI2 RSDP with the patched bootloder.

https://github.com/gifnksm/bootloader/commit/45b01fdefee9698ec412291904e19abe91d11c04

phil-opp, 2021-05-22:
On UEFI systems we could search for the ACPI v2 RSDP first and only look for a v1 RSDP if no v2 RSDP is found. The question is what we want to do on BIOS systems. Right now, we do an [RSDP search](https://docs.rs/rsdp/1.1.0/rsdp/struct.Rsdp.html#method.search_for_on_bios) in common memoy regions that returns the first RSDP tag found. Maybe it makes sense to make the search function an `Iterator` instead or at least accept a start offset. Then we could prioritize the v2 RSDP there too.

IsaacWoods, 2021-05-23:
Aside from the discussion in rust-osdev/acpi#92, I had another look at the ACPI spec and it does offer some guidance on the correct thing for us to do on UEFI:
> The OS loader for an ACPI-compatible OS will search for an RSDP structure pointer (RSDP Structure) using the current revision GUID first and if it finds one, will use the corresponding RSDP structure pointer. If the GUID is not found then the OS loader will search for the RSDP structure pointer using the ACPI 1.0 version GUID.
The OS loader must retrieve the pointer to the RSDP structure from the EFI System Table before assuming platform control via the EFI ExitBootServices interface. See the UEFI Specification for more information.

There's nothing similar in the BIOS section, so I think the current approach, which only expects there to be a single RSDP in the relevant areas, is correct.

phil-opp, 2021-05-24:
Thanks a lot for looking into this! Seems like updating the UEFI code is a good starting point. I'll look into it.

@gifnksm Could you try whether the BIOS implementation already gives you the right RSDP?

closed 2021-05-24 by phil-opp

gifnksm, 2021-05-24:
@phil-opp The BIOS implementation gives RSDP for ACPI v1.
I think this is the intended behavior because QEMU's SeaBIOS is too old to support ACPI v2 (according to @bjorn3).

phil-opp, 2021-05-24:
Thanks for checking! Looks like we need to try it on real hardware then. I'm also fine with leaving the BIOS implementation as it is for now until someone encounters an issue with it.