rust-osdev/bootloader #173: Support cross-compiling in `builder`

nschoellhorn, 2021-05-23:
I am working on my operating system on an Apple Silicon M1 MacBook (aarch64) from time to time and when I try to build the boot image, it always fails with the message `This crate only supports the x86_64 architecture.` because it uses the current architecture, of course. I ended up cloning the repo and adding the `TARGET` environment variable to the builder process to make it cross-compile for x86_64, which works for me but I'm not sure if that's the best way to do this.

Maybe the crate could get a config option for cross-compiling or something? Or always force a specific target? I'm not sure what issues might arise from this, so I thought I'd submit this issue first to discuss stuff.

If we can agree on a good solution, I'd be happy to submit a PR for that! :)

Thanks for your work!

phil-opp, 2021-05-24:
Thanks for reporting! Could you be more specific about the error that occurs and how you were able to fix it. I tried looking through your `bmos` repo and found [this line](https://github.com/nschoellhorn/bmos/blob/9d9f453a31fd98cbb9b2c6818dc43ada17a53c24/boot/src/main.rs#L30), but I'm not sure if that's what you meant.

nschoellhorn, 2021-05-24:
Oh well, sorry, thought it would be obvious 😅 

So what I get on my M1 MacBook is the error thrown here: https://github.com/rust-osdev/bootloader/blob/6df1629a86827f1c46e8e997daa557084e697bbc/src/lib.rs#L105

Actually, that line you found was one try, but that didn't work. What I ended up doing on my MacBook, is changing the `builder` alias in the bootloader crate to be `run --target=x86_64-apple-darwin --bin builder --features builder --quiet --` (instead of invoking without a specific target).

Now that I keep thinking about that, it doesn't really make sense to enforce the target architecture being x86 for the builder itself, does it? What should be x86 is the target that the builder builds *for*, right? And since the target for the bootloader is hardcoded anyways, I don't really see a point in the compile-time check for x86 architecture. Am I missing something?

phil-opp, 2021-05-24:
Thanks for clarifying! So using the `x86_64-apple-darwin` target works because it passes the architecture check and is still able to run on your M1 (via emulation)?

> Now that I keep thinking about that, it doesn't really make sense to enforce the target architecture being x86 for the builder itself, does it? What should be x86 is the target that the builder builds _for_, right?

Yes, it doesn't make sense to force an architecture for the `builder` executable. Seems like I forgot to adjust the check when I added the `builder`. 

> And since the target for the bootloader is hardcoded anyways, I don't really see a point in the compile-time check for x86 architecture. Am I missing something?

I want to prevent people from trying to use the `bootloader` crate for their ARM kernel. (The library part of the bootloader is also built for the kernel target.) So I would like to keep the check, but we can of course disable it when the builder binary is built. I'll look into it, thanks again for reporting!

nschoellhorn, 2021-05-24:
> Thanks for clarifying! So using the `x86_64-apple-darwin` target works because it passes the architecture check and is still able to run on your M1 (via emulation)?
> 
> >

Yes, exactly. The image is still built for x86 and then I run it with a custom-patched qemu-system-x86_64, which works well :D
Debugging is quite a hassle still, since there's no GDB for aarch64, but using the bundled GDB multiarch that comes with CLion does the job most of the time 😅

phil-opp, 2021-05-24:
I created a PR with a potential fix in #176. Could you try whether it fixes your problem @nschoellhorn?

nschoellhorn, 2021-05-24:
Works like a charm, thank you so much for the quick reaction!

phil-opp, 2021-05-24:
Awesome, thanks for testing!

PR #176 (2021-05-24): Don't check target architecture for builder crate to support cross-compiling
Fixes #173
closed 2021-05-24 by phil-opp