rust-osdev/bootloader #183: How could I use uefi-rs and bootloader together?

actuallyexeon, 2021-07-11:
I'm trying to print some text on the screen using UEFI but my project uses bootloader and AFAIK I don't think you can use a different function for uefi-rs other than efi_main, is there any way I can accomplish this somehow?

ethindp, 2021-07-11:
No, you can't. The `uefi_main`/`efi_main` function is the only way to get access to EFI boot services. The bootloader exits boot services *before* calling `kmain` so this is not possible.

actuallyexeon, 2021-07-11:
So how would I go about printing text on the screen or something like that without the uefi-rs crate

ethindp, 2021-07-12:
@actuallyexeon You need to use the framebuffer that the bootloader gave you. I'm not too sure on the maths required -- I can't really figure that out myself (but would love to know how) -- and though the OSDev wiki has [this article](https://wiki.osdev.org/Drawing_In_a_Linear_Framebuffer) I didn't consider it much help. But maybe you'll find it of use.

closed 2021-07-12 by actuallyexeon