rust-osdev/bootloader #187: Framebuffer Unclear

actuallyexeon, 2021-07-29:
I know it's possible to increase the brightness of pixels but how can you change the color, I see it's possible in logger.rs but way to hard to understand. Does anyone know how to use the framebuffer?

bjorn3, 2021-07-29:
With VGA text mode, the VGA bios is responsible for turning the color codes into full rgb colors. It seems to be possible to reconfigure this pallette using a specific port write: https://forum.osdev.org/viewtopic.php?f=1&t=23753

actuallyexeon, 2021-07-29:
idk if you understand but i'm talking about the uefi framebuffer in boot_info.rs not vga text mode

bjorn3, 2021-07-29:
I see. In that case I think changing the following two lines would be enough: https://github.com/rust-osdev/bootloader/blob/bc6c1bd62de697dc090ff8e7131cc36b91a531f4/src/binary/logger.rs#L130-L131

In the first case the array values are the red, green amd blue amount padded with a 0. In the second case it is first blue, then green, then red.

actuallyexeon, 2021-07-30:
how could i create a pixel though? that's more of what i'm asking about

bjorn3, 2021-07-30:
The code I referenced is what determines the color. The pixel is written at https://github.com/rust-osdev/bootloader/blob/bc6c1bd62de697dc090ff8e7131cc36b91a531f4/src/binary/logger.rs#L136-L137

The code is a bit harder to read due to the fact that `bytes_per_pixel` may differ depending on the exact format the pixel data is in (rgb,bgr,greyscale), but it basically writes the pixel data to the right offset in the `framebuffer` slice.

actuallyexeon, 2021-07-30:
so how could i adapt that code to buffer_mut() in boot_info.rs

ethindp, 2021-07-30:
I think that somebody needs to create a dedicated guide about
explaining this and how it works. Yeah, the OSDev folks have an
article about it, but I don't think another would hurt, particularly
related to this particular bootloader. It seems like I'm not the only
one tripping up on this issue.

actuallyexeon, 2021-07-31:
I found out how you can get it to work. I think I'll document it somewhere

SpanishPear, 2022-04-09:
@actuallyexeon  did you manage to write up that document?

actuallyexeon, 2022-04-09:
> @actuallyexeon  did you manage to write up that document?

What do you mean by that?

SpanishPear, 2022-04-09:
Ah I was referring to

> I think I'll document that somewhere

Did you end up documenting it

actuallyexeon, 2022-04-13:
> Ah I was referring to
> 
> > I think I'll document that somewhere
> 
> Did you end up documenting it

No, but there is a piece of code by @anellie here that you can look at for reference https://github.com/anellie/yacuri/blob/main/kernel/src/graphics/mod.rs and you can also look at this https://github.com/cherrydevs/cherrykernel/blob/master/src/video/mod.rs that I made a while back using anellies code

SpanishPear, 2022-04-15:
Thanks :partying_face:

asensio-project, 2022-08-07:
I tried to change the pixel colour via changing: 
PixelFormat::BGR => [intensity / 2, intensity, intensity, 0]
But I see the colour that I put but no letters only a big line.
How I can fix it?

ChocolateLoverRaj, 2025-02-26:
Why do u have an array with 4 elements if the pixel format is BGR (just 3 elements)?