rust-osdev/bootloader #188: Error compiling with "cargo bootimage"

spocino, 2021-07-30:
When trying to compile a project using bootimage, I get an error as such:

```
Error: An error occured while trying to build the bootloader: The `bootloader` dependency has not the right format: No `package.metadata.bootloader.target` key found in Cargo.toml of bootloader

(If you're using the official bootloader crate, you need at least version 0.5.1)

Caused by:
    The `bootloader` dependency has not the right format: No `package.metadata.bootloader.target` key found in Cargo.toml of bootloader
    
    (If you're using the official bootloader crate, you need at least version 0.5.1)
```

I am using bootloader version `0.10.6`, and the latest bootimage package.

NathanMcMillan54, 2021-07-31:
Try looking at [this](https://github.com/rust-osdev/bootloader#usage)

phil-opp, 2023-01-02:
The `bootimage` crate is no longer used with `v0.10` and `v0.11` of the bootloader. See also our migration guides: https://github.com/rust-osdev/bootloader/tree/main/docs/migration

closed 2023-01-02 by phil-opp