rust-osdev/bootloader #190: Failed to add interrupt handling on top of the basic example

trusch, 2021-08-10:
I tried to follow the blog_os blog up to chapter 6, but started from the basic example of this repo, so that I'm on the latest version of the bootloader crate. Everything works like a charm until after adding the gdt and Pic and I get a first timer interrupt. The handler is successfully called and calls IRET for the interrupt in the end. After that I get a double fault.

Could this be interacting with the serial output that I also setup like in the blog posts? Is there anything different while the bootloader stage that I need to consider?

I would be grateful for any thought on this. If needed I can push my current state and paste a link here.

Perhaps @phil-opp himself has an idea? :)

lassipulkkinen, 2021-08-17:
I hit this problem a while ago after upgrading my existing project to bootloader 0.10.0, and have just now managed to solve it (I discovered this issue because I was looking for it). Apparently, the `iret` fails because the `ss` (stack segment) register points to the descriptor at GDT offset `0x10`, which isn't a data segment. Previous bootloader versions have set that register to `0x00`, which (I think) is always valid in ring 0. (This probably should've been mentioned in the changelog...)

blog_os doesn't include a data segment in its GDT, so it won't reload `ss` after the `lgdt`, either (although perhaps it should). Consequently, the "visible" value of `ss` is invalid with respect to the new GDT, but the processor is actually using its "hidden" copy of the segment descriptor, which is still valid. However, the validity of the register *is* checked when `iret` tries to pop its value from the stack, resulting in the error.

It follows that either adding a data segment to the GDT and loading that into `ss` and `ds`, or just loading `0x00` instead, fixes the problem.

trusch, 2021-08-18:
@lassipulkkinen Thank you so much, I guess it would have taken ages to figure that out by myself. Indeed adding a kernel data segment and storing it in SS solved the issue.

closed 2021-08-18 by trusch