rust-osdev/bootloader #191: Printing to 0xb8000

vityafx, 2021-08-17:
Hello there. I have been following [the article](https://os.phil-opp.com/vga-text-mode/). Using GRUB bootloader I was able to print to the VGA text buffer at the address `0xb8000`, while using the bootloader crate version `0.10`, qemu simply crashes when I run `cargo krun` using the `simple_boot` example. Could you please explain this behaviour? Should it be that way or not? How do I print to this address?
Using the framebuffer seems a bit odd, as I, at the initial stage, would just like to print symple ascii characters. I noticed you are already doing that with the yellow text on the black screen using the framebuffer (presumably), so most likely you already have bitmap-encoded fonts ready to use with the framebuffer? Can you also provide these examples?

bjorn3, 2021-08-17:
The current edition of blog os is written for bootloader 0.9, which only supports BIOS. Bootloader 0.10 had to switch to framebuffer usage as UEFI doesn't support anything like the VGA text buffer once you exit the boot services. Only a framebuffer provided by GOP is supported. To allow the same code to run on BIOS, the BIOS version had to use a framebuffer too.

vityafx, 2021-08-17:
Thanks very much for explaining that!
And about the last part, are there any bitmap fonts available for use with the framebuffer? Can I simply reuse the same functions the` bootloader` uses for printing the text?

closed 2021-08-17 by vityafx

bjorn3, 2021-08-17:
The bootloader crate uses the font8x8 crate for text rendering: https://github.com/rust-osdev/bootloader/blob/0f7f78382a8021f81d66fb149ab98a441a47a6c0/src/binary/logger.rs#L109-L125 You can copy these functions, but you can't directly call the bootloader version I think.

vityafx, 2021-08-17:
> The bootloader crate uses the font8x8 crate for text rendering:
> 
> https://github.com/rust-osdev/bootloader/blob/0f7f78382a8021f81d66fb149ab98a441a47a6c0/src/binary/logger.rs#L109-L125
> You can copy these functions, but you can't directly call the bootloader version I think.

Yes, I noticed that.  :) Can the bootloader crate export this functionality with documentation? So, as a part of the library.